letscontrolit/ESPEasy #1772: Plugin template

jjansen85, 2018-09-21:
This line throws an error during compiling of a new plugin

> error: 'plugin_not_initialised' was not declared in this scope
>        if (plugin_not_initialised)
>            ^

https://github.com/letscontrolit/ESPEasy/blob/27a03f166aa43e974394d21fc3e8f40e0dcbbfef/src/_Pxxx_PluginTemplate.ino#L187

TD-er, 2018-09-21:
I think there are more things to update on the template.
But this line you mention is more 'pseudo code' to show what you must check for.
Every plugin has its own schema for this and also you must define global variables like these using something with the plugin name in it or else it will collide with other plugins.
Still that's not the best approach, since you then only can handle one instance of the plugin at the same time.

jjansen85, 2018-09-22:
Ah I aww, clear, thanks!

Grovkillen, 2018-09-22:
We close this then.

closed 2018-09-22 by Grovkillen