letscontrolit/ESPEasy #1774: Manual reboot () due to hardware watchdog in release mega-20180922

micropet, 2018-09-22:
The current version still has the same problem as the versions of the last few weeks.
I use ESP_Easy_mega-20180922_test_ESP8266_4096.bin

The units without sensors run only slightly longer than one or four hours.

Units with Sensors (BME280 BH1750 Pir MH-Z19 TVoc CSS811 PMS7003)
boot after a few minutes.
The running time is between 3 minutes and 30 minutes. It is always different.

TD-er, 2018-09-22:
Do they have all these sensors at the same node?

micropet, 2018-09-22:
Yes

TD-er, 2018-09-22:
I just flashed a node with the SDS011 with this firmware, to see if serial I/O may be causing these issues.
I will now flash another node with the BME280 and MH-Z19 to see if those are enough to get the same behavior.
I noticed the BME280 plugin sometimes has a long time logged in my statistics, so that may be one of the culprits. Maybe you could disable that one as a test to see if it improves stability.

micropet, 2018-09-22:
Because the PMS7003 still does not work, most units have an SDS021.
So:

BME280   BH1750 Pir MH-Z19 TVoc CSS811 SDS021

micropet, 2018-09-22:
So disable the BME280?
Good, I'll do that.

TD-er, 2018-09-22:
On the other hand, the reported reboot intervals you report are way shorter than anyone else.
So we should start to dig down somewhere.

Grovkillen, 2018-09-22:
Yes, power is always a tricky issue to evaluate. I use 5V USB UPS on some of my units and they never reboot.

So could you give us info about the setup you're using?

v-a-d-e-r, 2018-09-22:
My 16 Nodes are all running fine now with all current changes for 1 day and 16 hours. No single reboot! :-) And I have all kind of sensors in use with usage of GPIO, I2C and HW serial....

Grovkillen, 2018-09-22:
Please do, but if they all are reporting reboots it may very well be a network issue as well?

Grovkillen, 2018-09-22:
And just a curious question. Do they ALL have these capacitors? What if you remove those on one? As a test?

TD-er, 2018-09-22:
The power sounds OK to me.
Also those would not likely result in watchdog issues.
Maybe I could lower the core library to 1.7.3, since the number of reported watchdog reboots has increased a lot after the update to 1.8.0.
Not that they were not reported before, but the reports of those reboots is a lot more than before.

It could still be WiFi related.

micropet, 2018-09-22:
I do not believe in a network problem.
Currently, 51 WLAN devices are registered on both Unifi Access Points.

In the network are about 20 Wemos D1 with an old and simple, programmed by me software.

There are several LED drivers with 3-6 100 Watt LEDs connected to these units. I have been using this for years to switch the light in the apartment via PIR.

These units run for months without rebooting.

So, the same hardware I use for ESPEasy.

@Grovkillen
Yes. All have this capacitors, also my own Units.

TD-er, 2018-09-22:
What core lib do these other nodes use?
Could be 2.3.x or older even?

micropet, 2018-09-22:
@TD-er 

No idea. The version is already several years old.

Because they work, I have not changed anything.

TD-er, 2018-09-22:
I have just been looking at the uptime of my nodes.
One of them is running for 42 days now and is running ESP_Easy_mega-20180513_normal_ESP8266_4096.bin 

So I will look into what core lib that was.

micropet, 2018-09-22:
Now we have 2.4.2?

TD-er, 2018-09-22:
Yep, so simply changing to 2.4.1 in the platformio.ini could help.
I can make a build with that for you to try if you like.
What build version do you need? (normal/test and flash size)

micropet, 2018-09-22:
Thank you Gijs.
Is not necessary. I can change the core version myself and compile with platformio.

micropet, 2018-09-22:
It may be a coincidence, but the unit with the BME280 disabled has been running for 5 hours now.

micropet, 2018-09-22:
Yes, the PMSx003 plugin only worked for me for a few minutes.
Then no more data comes.

The BME280 is quite important, I think. With temperature, pressure and humidity, I find no alternatives to a good price.

TD-er, 2018-09-22:
It sure is and it is probably one of the more popular plugins.
So I will have a look at it, to see why it appears to take up-to 1.5 seconds sometimes. At least that's what my statistics claim

micropet, 2018-09-22:
Wau, thats much time.

TD-er, 2018-09-22:
Just one of such lines in my stats dump:
`5132780 : PluginStats P_27_Environment - BMx280 ONCE_A_SECOND        Count: 30 Avg/min/max 53395.60/306/1588655 usec`

So that is quite close to the (software) watchdog timeout and maybe the recent versions changed the hardware watchdog timeout to match the 2 sec.

micropet, 2018-09-22:
That may well be possible.
My unit is now running 6 hours 27 minutes without BME280. :)

Can you perhaps adjust something in the library (filter standby ...)?

TD-er, 2018-09-22:
I found the bug in the BME280 plugin.
Should be fixed in #1779

Just curious, is it still running fine with BME280 disabled?

micropet, 2018-09-22:
Booted an hour ago.
But it was about 7 hours online.

TD-er, 2018-09-22:
You could check the latest merge with BME280 enabled

ShardanX, 2018-09-22:
For testing here:
- 2 nodeMCU, cleaned and flashed with 20180922.
- Two "Raspberry" power supplies 5V 2A.
- first has only "Uptime" in the tasks.
- Second has Uptime and a BME220 attached.
On both NTP and UDP is set, no controller, no other things.

Now up and running for 6 hours, I'll watch and message again tomorrow.
Hope that helps.
Anything I should look at?

micropet, 2018-09-22:
> You could check the latest merge with BME280 enabled

Just flasht.

TD-er, 2018-09-22:
@ShardanX I guess the problems with the BME280 manifest with more devices connected which may send data to the node.
For example sensors sending data via serial without being requested for data.
The BME280 was blocking for 1.5 seconds. This could give issues when buffers are being filled and lots of interrupts triggered which could extend this blocking time even more.

ShardanX, 2018-09-22:
> @ShardanX I guess the problems with the BME280 manifest with more devices connected which may send data to the node.
> For example sensors sending data via serial without being requested for data.
> The [...]

Other sensors on I²C or on other interfaces?
Just added some more I²C and restarted both nodes

TD-er, 2018-09-22:
I2C devices are "pull", so you have to request them to send you data.
Some devices on serial also have a pull-like way of communicating (Modbus/Mbus), but others just push the data.
That last group may interfere with the processes running on the CPU and thus extend processing time, or cause buffers to overflow when not dealt with in due time.

s0170071, 2018-09-23:
Just a note: I had WD issues recently when I was setting the serial baudrate to 9600 with debug messages on... @micropet your sensors are all I2C?

micropet, 2018-09-23:
@s0170071 
Not all.
Only:
BME280   BH1750  CSS811

TD-er, 2018-09-23:
@micropet What is the "reboot interval" now, with the changed BME280 code?

micropet, 2018-09-23:
ca. 5 Stunden

TD-er, 2018-09-23:
So that's at least a factor 10 improvement from the start of this issue ;)
Is the reboot reason always the same (Hardware Watchdog), or do you also get other reasons?

micropet, 2018-09-23:
It is always the same: Hardware Watchdog
And yes factor 10 - 15 better.

TD-er, 2018-09-23:
Just a few of these iterations and we can call it stable, or at least very hard to reproduce  ;)

micropet, 2018-09-23:
very hard to reproduce - That's the way it is.

micropet, 2018-09-23:
A big advantage of espeasy I see in it, that after a reboot the unit is functional again.

This is not always the case with my own buildings. I had to briefly disconnect from the supply voltage so they worked again.

Maybe there will be time for things like:
PMS7003
or very important to me:
PCA9685 values invert
Currently, at a value of 0, the Led is at full power
at 4092 the led is off.

ShardanX, 2018-09-24:
FYI: Both of my nodes (see above) are running for about 48h without problem now.

micropet, 2018-09-24:
I have 7 reboots since this morning.

s0170071, 2018-09-24:
#1795  :-)

micropet, 2018-09-24:
Yes that would be nice.
1 or 2 days would be enough.  ;)

s0170071, 2018-09-25:
I found something in my rules that may have caused these frequent crashes. It was a comment with a leading space: 
```
on RelayOn do
 gpio,12,1
 // gpio,13,1
endon

```

There was a message in the logs about a  not recognized command, don't exacly remember what it was. Can check this evening though. Runtime was always less than 4 hours, since I removed the comment line the log message is gone and the unit is now approaching 12 hours.
Anyone else using rules ?

Grovkillen, 2018-09-25:
I use rules and comments but never on a single line and never with a space.... I must test!

Grovkillen, 2018-09-25:
I get this in the log but no crash:
```
465062 : Command:
465062 : Command unknown: ""
```

I use this rule to test:
```
on Rules#Timer=1 do
  Publish %sysname%/IP,%ip%
  Publish %sysname%/MAC,%mac%
  Publish %sysname%/AP,%bssid%
  Publish %sysname%/RCO2,[Carbondioxide#RCO2]% (%systime%)
  // TaskValueSet 4,1,2
  // gpio,13,1
  TimerSet,1,10
endon
```

s0170071, 2018-09-25:
Looks like what I have seen. I did not observe an immediate crash either. It may also be a coincidence that my unit is now surviving longer. 
Crash or not, it looks like its worth an issue.

TD-er, 2018-09-25:
Yesterday I made this commit: https://github.com/letscontrolit/ESPEasy/pull/1792/commits/bd2bdab95c102c700f4039540532858de8a30fa6
I tested it also on my ESP8266 and ESP32 and it seems to work.
What it does is quite simple, it moves the allocation of the large SettingsStruct from stack to heap.
My suspicion is that some of these crashes may happen due to stack overflow and moving such a large allocation on the stack to the heap may clear a lot of the very limited stack.

So maybe you could test that also on a number of nodes?

s0170071, 2018-09-25:
Maybe we should monitor the stack pointer ? This used to be good practice back in the days...

TD-er, 2018-09-25:
If you can give me a pointer (pun intended) on how to do that, that would be great :)

s0170071, 2018-09-25:
Someone made some experiments here:  https://www.esp8266.com/viewtopic.php?f=13&t=16734
There is even an issue open here: https://github.com/esp8266/Arduino/issues/5158 and https://github.com/esp8266/Arduino/issues/5148
The later one looks like there is already a method for checking the statck: getFreeContStack()

s0170071, 2018-09-25:
@TD-er: There is also a brute force approach: 

1. output a log message: "starting stack test..." , 
2. wait for it to be sent out (10ms) and then 
3. do 100? nested function calls.

If the stack is low this will crash the system. Thanks to the log message you know why.

s0170071, 2018-09-26:
@Grovkillen that unknown command was just a coincidence. The unit rebooted after 18 hours.

Grovkillen, 2018-09-26:
But might be some overload going on still?

TD-er, 2018-09-26:
Did anyone test my [suggestion](https://github.com/letscontrolit/ESPEasy/issues/1774#issuecomment-424233466) ?
I will merge that PR now, and it would be nice if someone could test it, before the nightly build is made.

N.B. when testing on ESP32, it will clear all your settings.

s0170071, 2018-09-26:
Isn't it too late for testing now ? You can tell only after a day or so if it is an success. 

A good indication could be the stack test (with the nested calls) like I suggested [here](https://github.com/letscontrolit/ESPEasy/issues/1774#issuecomment-424325847): 

As far as I know we have no clue whatsoever how full the stack is. I would really like to know because that would finally give a plausible explaination why ESPEasy stability is often so bad..

I will test the stack this evening or tomorrow but I can't promise. Im very low on time in the evenings :-(

giig1967g, 2018-09-26:
> 
> 
> Did anyone test my [suggestion](https://github.com/letscontrolit/ESPEasy/issues/1774#issuecomment-424233466) ?
> I will merge that PR now, and it would be nice if someone could test it, before the nightly build is made.
> 
> N.B. when testing on ESP32, it will clear all your settings.

Hi Gijs,
with latest releases, in my case the Hardware Watchdog is really random, some times the unit reboots after 2 hours, sometimes it doesn't reboot for days (now my units have been running for 2.5 days). So it's difficult to tell if your commit solves the issue. 
- 1 unit with 20180914 on for 6.5 days.
- 1 unit with BME280 and 20180923 on for 2.5 days
- 1 unit without BME280 and 20180923 on for 2.5 days
I am convinced that the HW WD problem is related also to network protocol delays.

Isn't it really possible to write catch the cause of the HW WD?

TD-er, 2018-09-26:
@giig1967g Did you test with that specific commit?
It was not merged until about an hour ago.
@s0170071 I know that these commits don't break the units, since it has been running on my nodes for a while now.
And the ESP32 merge should be done anyway, since with the default partitioning, you cannot add a lot with the latest core libraries. There was only a few kB left to add until building becomes impossible.

[124 more replies]

closed 2019-10-27 by TD-er