letscontrolit/ESPEasy #1775: Reconnect wifi roop in Mega-20180922

norimikan, 2018-09-22:
Use Sonoff RF and Mega-20180922 problem.

In case not flash blank_1MB.bin
after install Wifi is reconnect every 5sec.

In case flash blank_1MB.bin before install ESP_Easy_mega-20180922_normal_ESP8266_1024.bin
Can't see ESP_Easy_0 in WIFI.

Best Regards.

TD-er, 2018-09-22:
Did previous versions run fine on this node?
And can you get some serial logs to see what is happening?

norimikan, 2018-09-23:
Dear TR-er

serial log is

3135015: WIFI : Static IP: xxx.xxx.xxx.xxx (ESP-Easy-0) GW: xxx.xxx.xxx.xxx SN: 255.255.255.0 duration: 2 ms
3135027: EVENT: WiFi#Connected
3135051: EVENT: WiFi#Connected Processing time:24 milliSeconds
3135053: WIFI : Arduino wifi status: WL_CONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_SERVICES_INITIALIZED
3135128: MQTT : Intentional reconnect
3135129: Read settings: ControllerSettings index: 0
3135255: MQTT : Connected to broker with client ID: ESPClient_68:C6:3A:CD:A1:FC
3135256: Subscribed to: domoticz/out
3135258: EVENT: MQTT#Connected
3135283: EVENT: MQTT#Connected Processing time:25 milliSeconds
3135932: NTP : NTP host 0.asia.pool.ntp.org (118.140.184.99) queried
3136004: NTP : NTP replied: 70 mSec
3136006: EVENT: Time#Set
3136030: EVENT: Time#Set Processing time:24 milliSeconds
3143494: sendcontent free: 18936 chunk size:400
3143502: sendcontent free: 18936 chunk size:400
3143509: sendcontent free: 18936 chunk size:400
3143516: sendcontent free: 19064 chunk size:400
3143521: sendcontent free: 18976 chunk size:67
3143524: sendcontent free: 18304 chunk size:0
3144794: sendcontent free: 19992 chunk size:359
3144798: sendcontent free: 19992 chunk size:0
3146066: sendcontent free: 19992 chunk size:280
3146070: sendcontent free: 19816 chunk size:0
3147381: sendcontent free: 19992 chunk size:280
3147386: sendcontent free: 19816 chunk size:0
3148658: sendcontent free: 19992 chunk size:280
3148663: sendcontent free: 19992 chunk size:0
3149938: sendcontent free: 19992 chunk size:280
3149942: sendcontent free: 19992 chunk size:0
3151206: sendcontent free: 19992 chunk size:280
3151210: sendcontent free: 19992 chunk size:0
3151939: Memtrace^0: lowest: 13888 PluginCall_s (0)-> 17784 addToLog-> 17968 LoadControllerSettings-> 17784 LoadFromFile-> 17616 rulesP
3151940: 1: lowest: 14736 PluginCall_s (0)-> 17784 PluginCall_s (0)-> 17784 addToLog-> 17968 LoadControllerSettings-> 17784 LoadFromFil
3151941: 2: lowest: 13040 addToLog-> 17968 LoadControllerSettings-> 17784 LoadFromFile-> 17616 rulesProcessing-> 15872 rulesProcessingF
3151943: WD : Uptime 53 ConnectFailures 2 FreeMem 19760
3152486: sendcontent free: 19776 chunk size:400
3152498: sendcontent free: 19992 chunk size:175
3152502: sendcontent free: 19808 chunk size:0
3152829: LoopStats: shortestLoop: 76 longestLoop: 2935469 avgLoopDuration: 101.80 loopCounterMax: 394736 loopCounterLast: 276070 countFi
3152830: PluginStats P_1_Switch input - Switch ONCE_A_SECOND Count: 60 Avg/min/max 22.78/7/81 usec
3152832: PluginStats P_1_Switch input - Switch TEN_PER_SECOND Count: 582 Avg/min/max 30.48/8/99 usec
3152833: PluginStats P_1_Switch input - Switch FIFTY_PER_SECOND Count: 2876 Avg/min/max 19.93/7/61 usec
3152834: Load File stats: Count: 1 Avg/min/max 5272.00/5272/5272 usec
3152835: Plugin call 50 p/s stats: Count: 1438 Avg/min/max 971.52/850/1404 usec
3152836: Plugin call 10 p/s stats: Count: 291 Avg/min/max 944.81/874/1381 usec
3152838: Plugin call 10 p/s U stats: Count: 291 Avg/min/max 3422.83/3182/4137 usec
3152839: Plugin call 1 p/s stats: Count: 30 Avg/min/max 5782.50/1487/115775 usec
3152840: setNewTimerAt() stats: Count: 1891 Avg/min/max 132.32/114/248 usec
3152841: timeDiff() stats: Count: 1045172 - CPU cycles per call: 8.63
3152842: Scheduler stats: (called/tasks/max_length/idle%) 276070/1885/7/87.50
3153876: sendcontent free: 19776 chunk size:400
3153882: sendcontent free: 19776 chunk size:400
3153889: sendcontent free: 19776 chunk size:400
3153895: sendcontent free: 19776 chunk size:400

Best Regards.

TD-er, 2018-09-23:
You stated this "EVENT: WiFi#Connected" does occur every 5 seconds.
But this log covers almost 19 seconds and only 1 occurrence.

I see you're using static IP config.
Can you also try DHCP?
And can you give a screenshot of the IP configuration? (make sure all 4 fields are set)

norimikan, 2018-09-23:
Dear TD-er

I try to set DHCP(empty 4 fields), but cannot connect wifi.

static IP screanshot is this.
[image]

sorry for not attache log.
I can get only rsyslog from server.

Best Regards.

norimikan, 2018-09-24:
Dear TD-er

Now I use EspEASY_Mega_20180914/Sonoff products is work fine.

1.How often do you get reconnects and what is reconnecting (wifi/MQTT) ?
   Use Mega_20180922 wifi is disconnect and reconnect about every 10sec.

2.How strong is the wifi reception (RSSI value)
   -65dB

3.is any computer in your network able to get an IP-address via DHCP (via WiFi)?
   Mega_20180914 can connect dhcp.

4.Did this node work fine with older versions of ESPeasy, or maybe other firmwares?
   Use EspEASY_Mega old versions is work fine.(now use Mega_20180914)

5.Has this node been used in this network before?
   Use Sonoff RF and Sonoff 4ch pro use Esp_EASY_Mega in this network.
   Sonoff RF is use Mega_20180910 and Mega_20180914.
   Sonoff 4ch pro is use Mega_20180910.

6.When wifi connection is lost/reconnected, is the node rebooting?
   Not rebooting. But Wifi is lost and Reconnect.

7.What plugins are installed?
   Install Domoticz_MQTT

8.How did you manage to get the wifi connection setup, without an AP appearing?
   set wifi in install EspEasy first time.(Connect Esp_Easy_0 and set AP)

9.Do you have multiple accesspoints in the network with the same SSID?
   Only one access point.

10.Is there any plugin making use of the MQTT controller which is active?
   Use MQTT controller is Domoticz.

I try not set controller MQTT and get log is

162755: EVENT: WiFi#Connected
163485: EVENT: Time#Set
171911: EVENT: WiFi#Disconnected
171915: WIFI : Disconnected! Reason: '(200) Beacon timeout' Connected for 9172 ms
172360: WIFI : Connecting xxxxxx attempt #0
172361: IP : Static IP : 192.168.xxx.xxxGW: 192.168.xxx.xxx SN: 255.255.255.0 DNS: 192.168.xxx.xxx
172737: WIFI : Connected! AP: xxxxxx (9C:28:EF:B5:E0:94) Ch: 9 Duration: 375 ms
172738: IP : Static IP : 192.168.xxx.xxx GW: 192.168.xxx.xxx SN: 255.255.255.0 DNS: 192.168.xxx.xxx
172740: WIFI : Static IP: 192.168.xxx.xxx (ESP-Easy-0) GW: 192.168.xxx.xxx SN: 255.255.255.0 duration: 3 ms
172752: EVENT: WiFi#Connected
173475: EVENT: Time#Set
181893: WD : Uptime 3 ConnectFailures 0 FreeMem 18344
187361: EVENT: Clock#Time=Mon,17:52
193737: IP : Static IP : 192.168.xxx.xxx GW: 192.168.xxx.xxx SN: 255.255.255.0 DNS: 192.168.xxx.xxx
193738: WIFI : Static IP: 192.168.xxx.xxx (ESP-Easy-0) GW: 192.168.xxx.xxx SN: 255.255.255.0 duration: 3 ms
193750: EVENT: WiFi#Connected
194495: EVENT: Time#Set
203761: EVENT: WiFi#Disconnected
203765: WIFI : Disconnected! Reason: '(200) Beacon timeout' Connected for 10 s

Best Regads.

norimikan, 2018-09-28:
I try use Mega-20180927 and work fine.

But probrem in set "Wifi Status LED" to "GPIO-13".

Wifi Status LED set to GPIO-13 and reboot hardware,
cannot connect wifi.
Serial log is 

1901 : WD   : Uptime 0 ConnectFailures 0 FreeMem 25464
3402 : EVENT: WiFi#Disconnected
3416 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2865 ms
4346 : WIFI : Connecting (AP) attempt #1
4347 : IP   : Static IP : 192.168.xxx.xxx GW: 192.168.xxx.xxx SN: 255.255.255.0 DNS: 192.168.xxx.xxx
7208 : EVENT: WiFi#Disconnected
7221 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2860 ms
7346 : WIFI : Connecting (AP) attempt #2
7346 : IP   : Static IP : 192.168.xxx.xxx GW: 192.168.xxx.xxx SN: 255.255.255.0 DNS: 192.168.xxx.xxx
10207 : EVENT: WiFi#Disconnected

In case set Wifi Status LED is "None", work is fine.

Best Regards.

TD-er, 2018-09-28:
@norimikan Sounds like this issue? https://github.com/letscontrolit/ESPEasy/issues/1642

norimikan, 2018-09-28:
Dear TD-er

Yes. Maybe same problem issue #1642.

I try 2 hardware Sonoff RF.(update from Mega-20180910 and New one)

Mega-20180910 is set LED to GPIO-13and work fine 2 hardware.

Use Mega-20180927 and set LED to GPIO-13 after reboot,
LED is slow flashing(every2sec) same 2 device.

Today I try to Sonoff 4ch pro and report again.

Best Regards.

norimikan, 2018-09-30:
Dear TD-er

I try Install Mega-20180930 and work fine!
(Install to Sonoff RF)

Set Wifi LED to GPIO-13 and work fine.

Thankyou and Best Regerds.

TD-er, 2018-09-30:
Good to hear it is now fixed.
Or at least seems to be fixed, since I don't know what was the problem.
Only thing I changed which may cause crashes is to properly initialize variables.

norimikan, 2018-09-30:
Dear TD-er

I try install Sonoff 4ch Pro and work fine too.

feeling in use Mega-20180930 is good response in web interface.

this development is very wonderful. 

I would like to thank you once again.

TD-er, 2018-09-30:
OK, then I guess we can consider this one closed :)

closed 2018-09-30 by TD-er