letscontrolit/ESPEasy #1781: ESP32 - in test field, ampersand character cutting text to the right

nicoguido, 2018-09-22:
Tested with the assets R20100, mega20180914, mega20180922

### Summarize of the problem/feature request
when submitting a text with an ampersand character in it, the texte on the rigth of the "&", including the "&", will be removed. I did the test with Config->Unite Name and with Controllers->Generic HTTP->Controller Publish

### Expected behavior
The text with the ampersand character should be kept after submitting.

### Actual behavior
Examples :
- Unit Name : the text "test&esp" 
----> the result after submitting is "test" 
- Controller Publish : the texte "device=%sysname%&taskid=%id%&cmd=%valname%&value=%value%"
----> the result after submitting is "device=%sysname%"

TD-er, 2018-09-22:
Have you also tested this on ESP8266?

nicoguido, 2018-09-22:
No, I can't, I'm sorry.

nicoguido, 2018-09-22:
I'm sorry, I can't test on ESP8266.

nicoguido, 2018-09-22:
I'm using a Lolin D32

TD-er, 2018-09-22:
I just tested it with ESP8266 and my test node is now called "breadboard&test"
So it is apparently a problem for ESP32 on older builds at least.
I tested it on my ESP32 which is running core library 1.2.0 and it works.
[image]

nicoguido, 2018-09-22:
I'm sorry, butI don't understand well.

I've tested the 3 prebuild versions R20100, mega20180914, mega20180922.
I do an esptool erase_flash before flashing.
 
Do you mean it's not part of newer builds ?
If I'm using the ESP_Easy_mega-20180922_esp32dev.bin directly from the zip, 
what version of the core is it using ? how can check the version othe core?

TD-er, 2018-09-22:
tomorrow's build will be using 1.2.0 core lib.

nicoguido, 2018-09-22:
Thx ! I'll test.

(I did some research, and I have a better understanding now. :-) )

nicoguido, 2018-09-23:
no more problem with the ESP_Easy_mega-20180923_esp32dev.bin.
Thx

closed 2018-09-23 by nicoguido