letscontrolit/ESPEasy #1783: Adding ability to enable - disable tasks in rules on the fly.

xury77, 2018-09-23:
Hi. It's my first issue, so if I did wrong - very sorry. 
It would be great if you could switch tasks while running with rules or gpio
For example, switching between two tasks for an OLED display.

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2018-09-23:
Could you explain a bit more what you mean?
Perhaps by describing a use case.

Just taking the last line you wrote, do you want to add a command to the Framed OLED plugin to trigger a page change instead of looping with a set interval timer?

xury77, 2018-09-23:
I want use one OLED display with two task, The first one is a task normal Oled, and the second is a task framed.
Sometimes I want to switch those tasks using rules or momentary switch connected to gpio.
For example:
Task nr 8 is Display - OLED SSD1306
Task nr 9 is Display - OLED SSD1306/SH1106 Framed
Of course using both at same time is not possible, but ability to switch between will be nice.
E.q in rules
TaskControll,8,0 - disable task, and stop running
TaskControll,9,1 - enable task an run it.
I'm sorry if I could not explain my idea. My English is  so poor :)

TD-er, 2018-09-23:
The Framed OLED allows for a number of pages, so I understand you may want to show different information based on some event.
But why do you want to switch between OLED and Framed OLED plugin?
What does the OLED plugin have the framed one doesn't have?

xury77, 2018-09-23:
I use OLED display for displaying values of my sensors. 
The framed display plugins allows to use pages and is a nice feature, but sometimes I want to see my values at same time.  Framed screen has less space to fit all needed values for me at same time.
Second thing is maybe sometimes enabling and disabling tasks in rules will be usable for other applications.
Now I some times use a web page to switch it, but  I think if is possible to using web panel It will not be difficult to implement with rules.

TD-er, 2018-09-23:
I am not sure all plugins do support full enable/disable to actually allow to enable another plugin to use the same pins, or even the same plugin with different settings.
But I get the idea.

xury77, 2018-09-23:
I am aware of this. This feature should only be used after deep analysis.
You would have to specify what happens to the current state of the task after it is deactivated. I know it can cause problems. It would be an option for those who really know what they are doing.

TD-er, 2020-06-28:
This has been implemented, so can be closed

closed 2020-06-28 by TD-er