letscontrolit/ESPEasy #1785: Max number of "PLUGIN_VALUENAME"

stoffelll, 2018-09-23:
### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

Hi Guys,
I'm trying to write an additional plugin for the BME680 which is using the official BOSCH-Lib.
Therewith I'm able to retrieve the BOSCH-Indoor Air Quality (IAQ).

As I'm not so familiar and not such a good coder, I started with an existing pluging (P056) and modified it according the BOSCH Arduino-Code-Example. I was able to compile it successfully, but then I recognized a mixed up "PLUGIN_VALUENAME"-list. Means the first 5 items, which I defined, were labeled correctly in the "Devices"-overview and for the other 3, the label was just empty.
After I added the label inside of the plugin, 3 other labels become empty.

In addition in one formular field it is always appearing "Temperature".

I'm testing it with the codebase from 2018-09-22

### Expected behavior

All defined Plugin_Valuenames should be displayed correclty.

### Actual behavior

Some PLUGIN_VALUENAMES are shown empty, even they were defined.

### Steps to reproduce

I'm using this code:
#define PLUGIN_300
#define PLUGIN_ID_300         300
#define PLUGIN_NAME_300       "Environment - BME680 with IAQ by Bosch"
#define PLUGIN_VALUENAME1_300 "Temperature"
#define PLUGIN_VALUENAME2_300 "Humidity"
#define PLUGIN_VALUENAME3_300 "Pressure"
#define PLUGIN_VALUENAME4_300 "Gas"
#define PLUGIN_VALUENAME5_300 "IAQ"
#define PLUGIN_VALUENAME6_300 "RawTemperature"
#define PLUGIN_VALUENAME7_300 "RawHumidity"
#define PLUGIN_VALUENAME8_300 "AccuracyIAQ"

### System configuration

Hardware: NodeMCU v3

ESP Easy version: 2018-09-22

ESP Easy settings/screenshots: 
[image]
[image]

### Rules or log data

```

```

TD-er, 2018-09-23:
A plugin can only handle 4 values. 
See the define of `VARS_PER_TASK` in the `ESPeasy-globals.h`
If you want to use more, you either have to make a selection option in the plugin, or maybe split the plugin in parts.
Other option is to send events and catch them in rules if you really need those values.

I guess only temperature, humidity, air pressure and the IAQ values are really needed.

When forcing more than 4 values, you will see undefined behavior and very likely cause data corruption.

stoffelll, 2018-09-23:
Thanks for that info! 

Is it working to increase the number to 8 in ESPeasy-globals.h?

Unfortunately at least the IAQ-estimation/accuracy would be helpful, as the sensor need some calibration time. While it is in calibration mode, I just get the value 25 for the IAQ....sure I can also filter this value 25 to see, if it is in calibration or not.

Thanks a lot

Grovkillen, 2018-09-23:
Can't you just have the plug-in trigger some events like @TD-er suggested and pass along some values that can be fetched using the %eventvalue%?

You can trigger different events for those values that didn't fit in the plug-in values list.

closed 2018-09-30 by stoffelll