letscontrolit/ESPEasy #1786: RTC.bootFailedCount should not be increased in case of a wanted reboot

v-a-d-e-r, 2018-09-23:
RTC.bootFailedCount increases in every case of a reboot, but should only increase in case of a unwanted reboot (crash).

TD-er, 2018-09-23:
This value should only increase when a boot is not complete.
Its value will be set to 0 after wifi connection has been established.
Maybe we should also clear that value after N seconds of run time?

v-a-d-e-r, 2018-09-23:
Or something like this...

String Command_System_Reboot...
  RTC.bootFailedCount = 0;
  saveToRTC();
.
.
.

TD-er, 2018-09-23:
I looked into the source code, but the only places where that value is being altered is here:
```c++
  bool firstLoopConnectionsEstablished = checkConnectionsEstablished() && firstLoop;
  if (firstLoopConnectionsEstablished) {
     firstLoop = false;
     timerAwakeFromDeepSleep = millis(); // Allow to run for "awake" number of seconds, now we have wifi.
     // schedule_all_task_device_timers(); Disabled for now, since we are now using queues for controllers.

     RTC.bootFailedCount = 0;
     saveToRTC();
  }
```

and here:
```c++
  //warm boot
  if (readFromRTC())
  {
    RTC.bootFailedCount++;
    RTC.bootCounter++;
    readUserVarFromRTC();
```

So I really don't see how this counter could keep counting, unless you're not connecting to the network, or give a command to reboot before the network connection is being made.
Is that what has to change?

closed 2018-09-23 by v-a-d-e-r

TD-er, 2018-09-23:
Should I change something in the main version?