letscontrolit/ESPEasy #1787: MQTT import device : autostart on reboot

sglbt, 2018-09-23:
Hi, 

Here is the problem I'm facing. 
Just need to know if it is an expected bevahior.

### Summarize of the problem/feature request
The device Generic : MQTT Import is disabled after each reboot

### Expected behavior
After reboot the device Generic : MQTT Import, should be enabled

### Actual behavior
After reboot the device Generic : MQTT Import, should is disabled and need to be re-enabled manually

### Steps to reproduce

1. Enable the device Generic : MQTT Import
2. Reboot the ESP
3. Check that the device is disabled

### System configuration

Hardware: SONOFF DUALL R2 - ESP 8266

ESP Easy version: mega-20180922

Thanks
SG

Grovkillen, 2018-09-23:
If you look in the log I suspect that the MQTT import is making the unit crash on boot. We are disabling plugins that are causing a crash.

v-a-d-e-r, 2018-09-23:
This plug-in make no problems here. The last changes makes it even better! CPU load dropped about 15% :-)

TD-er, 2018-09-23:
It has been fixed yesterday, so if you use last night's build, it should be fine :)
I close this one, since it has been fixed (yesterday)

closed 2018-09-23 by TD-er

sglbt, 2018-09-25:
Sorry I didn't thank you for your help. 
Everything works perfectly with the last update.

Thanks again.
SG