letscontrolit/ESPEasy #1788: Feature reuest: some sleep (light sleep, modem sleep, ...)

s0170071, 2018-09-23:
I noticed that the ESP draws quite some current during operation. About 80 mA which results in approx 0,5W power consumption. My 10 ESPs would then require 43kWh during a year.

There are several sleep states to prevent this. Best known is the deep sleep but this prevents the unit from being reachable via the web interface. 

But there are a couple of other sleep states: 
https://blog.creations.de/?p=149

Some of then can cut the power consumption to 30%.... Thats also good if you use a temperature sensor. The dissipated heat from the ESP gave me wrong readings in the past too.

Since the ESP is idling most of the time, there should be some sleep strategy implemented.

TD-er, 2018-09-23:
Up-to the first bracket in the subject line is very high on my to-do list :)

s0170071, 2018-09-23:
did some experimenting and found that a delay(1) helps a lot.

Just for testing I did insert  at the end of backgroundtasks(): 
 ```
 yield();
  statusLED(false);
// new
 if (Settings.WireClockStretchLimit) wifi_set_sleep_type(MODEM_SLEEP_T);
 if(Settings.WireClockStretchLimit <100)      delay(Settings.WireClockStretchLimit);
// end new
runningBackgroundTasks=false;
```

If I now set the clockstrechlimit to 1 I see the power of my Sonoff S20 drop from a constant 0.7W to a fluctuating 0.3-0.5W. 
The Arduino core version I used was the latest git. Info page shows ESP82xx Core d8a7a34c, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3

s0170071, 2018-09-23:
This should be configurable and only kick in after those 30s. I think the challenge is to keep it all stable.

That delay that I mentioned above does as far as I can tell not affect the wifi connection. The unit is still reachable.

TD-er, 2018-09-23:
That delay will cause a lot of issues when plugins need to run frequently.
For example the switch plugin.
I am trying to get rid of all uses of `delay` in the code, to avoid timing issues with plugins.

s0170071, 2018-09-23:
Just check out https://blog.creations.de/?p=149 There it says

> Note: Auto Modem Sleep will not happen if there are no delay() or only delay(0) calls in the code.

I think that's exactly what we see here. Plugins run no more often than 50 times per seconds. I was thinking of idling with delays if there is time left after all plugins ran. E.g. delay(millis()%20) or so.

TD-er, 2018-09-23:
The scheduler knows how much time is there to 'delay'.
It is just the next timestamp in the queue.