letscontrolit/ESPEasy #1793: EasyGlobalSync doesn't work

szperix, 2018-09-24:
Inter ESP network doesn't work. Disappeared option "Global sync" per task. In advanced options also lack of this option. All units in node List are visible but task doesn't create on remote unit when "submit" clicked on local unit.
Build: ESP_Easy_mega-20180822_normal_ESP8266_4096.bin

TD-er, 2018-09-24:
The global sync has been moved to a new controller (Controller 13)
So the sending and receiving side must have that controller enabled and the plugin you want to 'sync' to another node must have that controller set checked.
Also the same position on the other node must be empty.

szperix, 2018-09-24:
Why this feature is disabled by default now? Where I can enable it?

Grovkillen, 2018-09-24:
Its a controller (as it should be) and you activate it just as you would do any other controller. And then you publish to it by checking the task to publish to that controller number.

szperix, 2018-09-24:
I understand. Thanks very much for a help. I will test this afternoon. I suggest small update [wiki](https://www.letscontrolit.com/wiki/index.php/EasyGlobalSync) for this feature.

Grovkillen, 2018-09-24:
Yes agree, we're a bit behind on wiki updates.