letscontrolit/ESPEasy #1795: 120 days ;-)

s0170071, 2018-09-24:
@TD-er congrats! Nice Job. 
If I only knew which version that was...

System Info |  
-- | --
Unit: | 7
Uptime: | **120 days** 21 hours 34 minutes
Load: | 13% (LC=18312)
Free Mem: | 17464 (10048 - sendWebPageChunkedData)
Wifi RSSI: | -65 dB
Wifi Type: | 802.11N
IP: | 192.168.1.23
GW: | 192.168.1.1
Build: | 20000  - Mega
GIT version: | (custom)
Plugin sets: | [Normal]
Number of Plugins: | 46
Core Version: | 2_4_0_RC2
Flash Size: | 4096 kB
Flash Writes (daily/boot): | 0 / 17
Sketch Size/Free: | 574 kB / 2496 kB
Boot cause: | Manual reboot
Warm boot count: | 8
STA MAC: | a0:20:a6:15:89:6d
AP MAC: | a2:20:a6:15:89:6d
ESP Chip ID: | 1411437
Flash Chip ID: | 1458400

s0170071, 2018-09-24:
P.s. with sht25 plugin and mqtt enabled

Grovkillen, 2018-09-24:
[image]

TD-er, 2018-09-24:
Maybe we just should put a "release" label on that one from March?
It took a while due to the test cycle...

s0170071, 2018-09-24:
@td-er Either that or keep a list / issue with known good versions. Maybe add a column with possible issues....

papperone, 2018-09-25:
I can confirm on March release; I still use those ones in the modules I don't want to risk issues with latest build and I need to say they never reboot unlees I do it intentionally or for a power outage :)

giig1967g, 2018-09-25:
Also for me March releases are rock solid. (before Event based wifi)
But I have May releases that have been working for months with router ASUS and the same release doesn't work for more than 24 hours with router Mikrotik.
I believe that the issue is the networking.

flexiti, 2018-09-25:
The worst thing is that every update I have to run to the device and using RS232 to bring them to life ;-)
According to the suggestion, I will switch to the issue of March..but as we all switch to this edition, no one will test the current ... paragraph 22 :-)

And seriously ... I think that special attention must be paid to remote update, if it works, the pain is less ... you can always go back to the older version without opening the device when it is in a hard-to-reach place.

TD-er, 2018-09-25:
We currently have the option to disable plugins/controllers/notifications if they cause crashes.
Also I noticed myself that disabling plugins may help to OTA an update. (MH-Z19 sensor on GPIO pins that affect flash mode at boot)

flexiti, 2018-09-25:
> We currently have the option to disable plugins/controllers/notifications if they cause crashes.

what exactly does it mean?

TD-er, 2018-09-25:
@flexiti About a week ago, I added a check to count the number of unsuccessful boots and disable one-by-one the plugins/controllers/notifications.
This will stop as soon as there is a successful boot and thus leave the troubling item disabled.

This will allow the user to regain control when a plugin or configuration is causing crashes, like MQTT import did last week.

edgiru, 2018-11-12:
158 days?
esp32 or esp8266?

Grovkillen, 2018-11-12:
ESP8266, it rebooted after 186 days.

Muller59, 2019-10-09:
on a simple sonoff basic.....

[image]

good work, guys

closed 2019-10-10 by Grovkillen