letscontrolit/ESPEasy #1799: RSSI and SID

flexiti, 2018-09-25:
System Info - in addition Wifi rssi, it would be good to see which network is in use  (SSID normal or fallback ,or name of network)

TD-er, 2018-09-25:
It is showing the BSSID and the network name.
For example:

STA MAC | A0:20:A6:19:2A:BF
-- | --
AP MAC | A2:20:A6:19:2A:BF
SSID | **Lurch4 (34:31:C4:B1:8D:C7)**
Channel | 1
Connected | 21h14m
Last Disconnect Reason | (1) Unspecified
Number reconnects | 0

flexiti, 2018-09-26:
[image]

Grovkillen, 2018-09-26:
Click "more info"

Grovkillen, 2018-09-26:
By the way, I see no Git Version. Are you running self compiled FW?

TD-er, 2018-10-11:
Will be fixed when #1894 is merged.

closed 2018-10-11 by TD-er