letscontrolit/ESPEasy #1806: JSN-SR04T doesn't work

Fusseldieb, 2018-09-27:
### Summarize of the problem/feature request

I have tried the ESPEasy firmware and was quite amazed when I tried it out, but at the same time a bit frustrated. The ultrasonic sensor named JSN-SR04T doesn't work with the newest build (mega-20180927). Some older build that I've found on the internet works just fine, but it's hella outdated. (https://www.letscontrolit.com/forum/viewtopic.php?t=3432 - Last post). 
The recent build just makes the ultrasonic sensor 'tick' once at boot and ... yeah ... that was it.

It would be nice if someone could look into this, I (and probably many others) would be very glad.

### Steps to reproduce

Add a Ultrasonic sensor and it doesn't work. Returns zero every time. The log spils out that it can't read the value every second.

TD-er, 2018-09-27:
It is a duplicate of this one: https://github.com/letscontrolit/ESPEasy/issues/1518
You could try build 20180606.
It seems there are a lot of users of this plugin, so this one definitely needs more attention.

closed 2018-09-27 by TD-er

Fusseldieb, 2018-09-27:
> You could try build 20180606.
Downloading 20180606 worked. So at least a newer version exists where this is working...
But yeah, I would love to see this in a newer version too :)

Fusseldieb, 2018-10-21:
@TD-er Could you please leave this issue open, as it still is a issue. With it closed I'll never know if this is fixed in the "latest" version or not.

Thank you.

TD-er, 2018-10-21:
It should be fixed, as far as I know.
Have you tried last nights build for example?

Fusseldieb, 2018-10-21:
Sorry that I deleted my previous post, I didn't knew that you were online and if it the confusion on my part was appropriate.

I'll try to download the [latest build](https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20181021) and try. 

Thank you for your fast response.

TD-er, 2018-10-21:
No problem :)

Fusseldieb, 2018-10-21:
@TD-er Perfect. It's confirmed working in the latest build (20181021_normal) and it's a lot more accurate than before.

Before: 88.0 ... 89.0 ... 88.0 ... etc
Now: 88.12 ... 88.14 ... 88.13 ... etc

Just a few sitenotes if someone stumbles upon this post:
After the update via the GUI provided by ESPEasy (or just flashing it locally over USB), **you must re-add all devices** manually again, **or they won't work**. They weren't working initially when it rebooted.

TD-er, 2018-10-21:
OK, good to know.
From what version did you upgrade?

Fusseldieb, 2018-10-21:
I previously was on `20180606_normal`, like you suggested me some weeks earlier.

> You could try build 20180606.

It was crashing a lot and I'm really relieved that it does work now on a new version :)

TD-er, 2018-10-21:
Glad to see we made some improvement in stability