letscontrolit/ESPEasy #1808: ESP32 PWM flicker after the 20180922 release

redskinhu, 2018-09-27:
### Summarize of the problem/feature request

I have a RGB led strip on the ESP32 and its working fine with the mega-20180922 release but the releases after that causes flicker.

### Steps to reproduce

1. Connect the led strip to R: GPIO17, G: GPIO18, B: GPIO19

### System configuration

Hardware: ESP32

ESP Easy version: mega-20180922 is ok but the releases after causes flicker (official and self compiled too)

TD-er, 2018-09-27:
So, with core 1.2.0 it is fine and with core 1.3.0 this occurs.
Maybe it is also related to the similar reports on ESP8266, when updating the PlatformIO core library from 1.7.x to 1.8.0

redskinhu, 2018-09-27:
Yes that was my thought too.

redskinhu, 2018-10-09:
Hello

Release mega-20181008 is tested - still flickers.

Thx
T

TD-er, 2018-10-09:
We're still using 1.4.0 core library, so it is expected not to be fixed at this moment.
I've looked a bit into the changes (based on ESP8266) and it seems it is an intended change, but then we have to look into how the PWM functionality now must be called.

redskinhu, 2019-01-16:
Hi

Sorry i forgot to report:
I'm using 20181120 self compiled fw. No flicker.

Thx
T

closed 2019-01-16 by redskinhu

TD-er, 2019-01-17:
@redskinhu Just to be sure, this issue is about ESP32, you're running ESP32 too?

And if so, do you experience crashes of the node when accessing certain web pages?