letscontrolit/ESPEasy #1810: logging race condition

s0170071, 2018-09-27:
seen today:

s0170071, 2018-09-27:
I was using debug-dev and had a lot of output. I also had rules enabled and set them in a way that the rule would toggle a gpio pin if a switch (RelayState) was triggered. Mean like I am I configured the switch to the same pin that the rule triggered. Just to annoy the ESP.

I am still looking for a rather quick and reliable way to crash it- for debugging purposes of course!

```
on RelayOn do
 gpio,12,1
gpio,0,1
endon

 

on RelayOff do
 gpio,12,0
gpio,0,0
// gpio,13,1
endon

on toggle do
if [RelayState#RelayState]=1
 event,RelayOff
else
 event,RelayOn
endif
endon

```

TD-er, 2018-09-27:
> I am still looking for a rather quick and reliable way to crash it- for debugging purposes of course!

Sure you do ;)
Without looking at the code, it could be I already increase the read pointer, while still reading from the buffer to output it to the serial port.
It should be increased after reading and the write should increase it before starting to write.
Maybe there is something there.

TD-er, 2018-09-27:
Those state messages are not sent via the buffer.
Those are from the main library's debug and are not sent to the buffer I added.
I have no idea how that can be grabbed and stored in the buffer.
Maybe I should stop sending from the buffer as soon as I see a newline.
Then each line is somewhat an atomic operation.

s0170071, 2018-09-27:
Maybe they are written to serial IF directly, byte by byte when you yield() ?

TD-er, 2018-09-27:
See `void addToLog(byte logLevel, const char *line)` in `Misc.ino`

s0170071, 2018-09-27:
no, its not there. It must be here:

TD-er, 2018-09-27:
OK, then we could run into a bug at another level.
I have seen something similar when running the web-log viewer at maximum throughput.
You can see other data being mixed with the log outputs. Look for "Bugje" in the code to find my tracer for when I was looking into that issue.
Never fixed it though.

s0170071, 2018-09-28:
I think I know whats going on. 
You write out the serial buffer only as far as there is room left in the Uart tx buffer. Thats ok, but when you stop because the buffer is full, other serial.print commands can kick in and place a byte in the stream. Then they stop because the buffer is full and the cycle repeats. 

I inserted two flush commands that seem to fix the issue. I am aware that this seems to slow down the system in certain circumstances. But this is only apparently. When you flush() you wait, but then the buffer is empty again and other serial.print commands will not have to wait.

 I will continue testing.

```
void process_serialLogBuffer() {
  if (serialLogBuffer.size() == 0) return;
  if (timePassedSince(last_serial_log_emptied) > 10000) {
    last_serial_log_emptied = millis();
    **Serial.flush();**
    serialLogBuffer.clear();
    return;
  }
  size_t snip = 128; // Some default, ESP32 doesn't have the availableForWrite function yet.
#if defined(ESP8266)
  snip = Serial.availableForWrite();
#endif
  if (snip > 0) last_serial_log_emptied = millis();
  size_t bytes_to_write = serialLogBuffer.size();
  if (snip < bytes_to_write) bytes_to_write = snip;
  for (size_t i = 0; i < bytes_to_write; ++i) {
    Serial.write(serialLogBuffer.front());
    serialLogBuffer.pop_front();
  }
  **Serial.flush();**
}
```

TD-er, 2018-09-28:
Maybe we can lower the number of bytes to write, or stop at a line break, depending on the set baudrate.
For example allow to write up-to N msec at once.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er