letscontrolit/ESPEasy #1811: serial commands available even when the password is set

jimmys01, 2018-09-27:
### Summarize of the problem/feature request

Set password protection for the ESP
go to serial and type "settings" or "reset"
esp gives out all the wifi passwords no questions asked.

### Expected behavior

Ask for the password, at least for the sensitive commands that reveal passords.

giig1967g, 2018-09-27:
I think that the reason is because the serial interface is not an SSH protected access.

jimmys01, 2018-09-28:
Is there a reason why someone with physical access can grab the wifi password just by typing "settings".

Grovkillen, 2018-09-28:
No that shouldn't be the case. We should ask for admin password for that.

uzi18, 2018-09-28:
just add password parameter like:
settings,password
if password is setted and compared ok, show data

uzi18, 2018-09-28:
or add cmd: unlock, password
to temporary unlock password, for example to reboot