letscontrolit/ESPEasy #1812: Setting email subjects within rules

Bejobe, 2018-09-27:
### Summarize of the problem/feature request

In the moment (mega-20180927) it is not possible to set email subjects within the rules. To have different email subject you have to define different notifications. But there are only 3 of it...,    

### Expected behavior

Example for a rule:

On SendEmail do
   notify %eventvalue%, "{{Subject here: %eventvalue%}}  Uptime: %uptime%m"
endon

### Actual behavior

Only 3 (static!) subjects in 3 notifications.

### Steps to change
 
Please change in file _N001_Email.ino the single line 113 from:
		mailheader.replace(String(F("$subject")), aSub);

to the following 12 lines (113 - 124):
		int sub_left  = aMesg.indexOf('{{');
		int sub_right = aMesg.indexOf('}}');
		if (sub_left == -1 || sub_right == -1) {
		    mailheader.replace(String(F("$subject")), aSub);
		} else {
			String newSub = aMesg.substring(sub_left, sub_right+1);
			aMesg.replace(newSub, "");
			newSub.replace("{{", "");
			newSub.replace("}}", "");
			newSub.trim();
		    mailheader.replace(String(F("$subject")), newSub);
		}	

Thank you very much for your nice and fine very useful software and with best regards 
Bejobe

### System configuration

Hardware: NodeMCU esp8266

ESP Easy version:  mega-20180927

TD-er, 2018-09-27:
As far as I know, this syntax with '{{' is not used anywhere in the software.
Such changes should be more like other commands.

Bejobe, 2018-10-04:
Thank you for your answer. 
It is only important, that it is possible, to set the subject within the rules together with the email text. 
Please change any other part of my request (seperator, implementation, ... ) in the way you think about.

closed 2022-04-22 by TD-er