letscontrolit/ESPEasy #1814: possible DoS attack + crash eover serial

s0170071, 2018-09-27:
current build; in Serial.ino 
again, no one is feeding the poor little watchdog :-(
Now, if you flood the serial with bytes, the dog will bite you.
```
void serial()
{
  while (Serial.available())
  {
    yield();
 
...
  }
}
```

TD-er, 2018-09-27:
As far as I know, calling `delay` or `yield` will also feed the watchdog.
Unless that has changed in some more recent version of the core libs????

This function is called from either the call to backgroundtasks or on ESP32 when RTOS flag is enabled, then it is executed in another "thread". (RTOS calls them "Tasks")

s0170071, 2018-09-27:
Ah, you're right. yield does also feed the dog. Since it does not allow for modem sleep I assumed it would not feed the dog either.

closed 2018-09-27 by s0170071