letscontrolit/ESPEasy #1816: sendcontent free: 17576 chunk size:0

s0170071, 2018-09-27:
when enabling log debug dev for serial and log debug for web, 
open the weblog, I see on the serial log: 

```
442493 : sendcontent free: 17576 chunk size:0
443522 : sendcontent free: 17496 chunk size:118
443526 : sendcontent free: 16976 chunk size:0
444563 : sendcontent free: 17576 chunk size:118
444566 : sendcontent free: 17576 chunk size:0
445602 : sendcontent free: 17576 chunk size:118
445606 : sendcontent free: 17576 chunk size:0
446629 : sendcontent free: 17576 chunk size:118
446632 : sendcontent free: 17576 chunk size:0
447665 : sendcontent free: 17576 chunk size:118
447668 : sendcontent free: 17576 chunk size:0
448696 : sendcontent free: 17576 chunk size:118
448700 : sendcontent free: 17056 chunk size:0
449729 : sendcontent free: 17576 chunk size:118
449733 : sendcontent free: 17576 chunk size:0
450734 : UDP  : 5C:CF:7F:39:B7:B0,192.168.1.24,8
450763 : sendcontent free: 17576 chunk size:118
450767 : sendcontent free: 17576 chunk size:0
451794 : sendcontent free: 17576 chunk size:118
451797 : sendcontent free: 17576 chunk size:0

```

Is this normal ?
Whats that chunk size 0: log entry ?

TD-er, 2018-09-27:
That's a left over debug for the WebServer changes we made a while ago.
The reason our web page is so fast is the chunked data transfers.
To end a chunked transfer, you send a chunk with size 0.

Grovkillen, 2020-01-10:
I close this one

closed 2020-01-10 by Grovkillen