letscontrolit/ESPEasy #1818: Soft WDT reset on rules save

s0170071, 2018-09-28:
ESP8266, current core and ESPEasy
Steps to reproduce: 

use these rules: 
```
on System#Boot do
 event,toggle
endon

on RelayOn do
event,toggle
endon
 

on RelayOff do
event,toggle
endon

on toggle do
if [RelayState#RelayState]=1
 event,RelayOff
else
 event,RelayOn
endif
endon

```

These are sort of oscillating rules. They will self-trigger. In the webinterface you can see the memory draining. 
Wait some seconds (memory is still well above 10k) and comment out the toggle commands in the rules. 
Hit the save button -> done.

Exception decoder shows: 
```

Decoding 34 results
0x40257aa0: String::~String() at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 735
0x40234014: parseCompleteNonCommentLine(String&, String&, String&, bool&, bool&, bool&, bool*, bool*, unsigned char&, unsigned char&) at /home/john/Arduino/scetchbooks/ESPEasy/WebServer.ino line 5798
0x40203e8b: SPIFFS_lseek at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/spiffs/spiffs_hydrogen.c line 1122
0x40203c11: spiffs_hydro_read at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/spiffs/spiffs_hydrogen.c line 1122
:  (inlined by) SPIFFS_read at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/spiffs/spiffs_hydrogen.c line 427
0x4025929c: SPIFFSFileImpl::position() const at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/spiffs_api.h line 352
0x4025ae81: fs::File::available() at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/FS.cpp line 153
0x40235a74: rulesProcessingFile(String, String&) at /home/john/Arduino/scetchbooks/ESPEasy/WebServer.ino line 5798
0x4025b688: getSpiffsMode(fs::OpenMode, fs::AccessMode) at ?? line ?
0x40257b00: String::reserve(unsigned int) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 735
0x40257b45: String::copy(char const*, unsigned int) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 735
0x40257c48: String::operator=(String const&) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 735
0x40235bef: rulesProcessing(String&) at /home/john/Arduino/scetchbooks/ESPEasy/WebServer.ino line 5798
0x40257a84: String::invalidate() at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 735
0x40236303: return_command_success at /home/john/Arduino/scetchbooks/ESPEasy/WebServer.ino line 5798
:  (inlined by) Command_Rules_Events(EventStruct*, char const*) at /tmp/arduino_build_817154/sketch/Commands/Rules.h line 34
0x40232ed0: doExecuteCommand(char const*, EventStruct*, char const*) at /home/john/Arduino/scetchbooks/ESPEasy/WebServer.ino line 5798
0x4024f787: WiFiUDP::parsePacket() at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiUdp.cpp line 282
0x4023406a: checkUDP() at /home/john/Arduino/scetchbooks/ESPEasy/WebServer.ino line 5798
0x40257ac7: String::changeBuffer(unsigned int) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 735
0x40100579: _umm_realloc at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1503
:  (inlined by) realloc at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1731
0x40257ac7: String::changeBuffer(unsigned int) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 735
0x4010020c: _umm_free at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1299
0x40257ac7: String::changeBuffer(unsigned int) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 735
0x40257b92: String::String(char const*) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 735
0x40106a62: millis at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.c line 183
0x4023bfb3: process_system_event_queue() at /home/john/Arduino/scetchbooks/ESPEasy/WebServer.ino line 5798
0x4020e08a: statusLED(unsigned char) at /home/john/Arduino/scetchbooks/ESPEasy/WebServer.ino line 5798
0x40201e4e: delay at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.c line 54
0x40234418: backgroundtasks() at /home/john/Arduino/scetchbooks/ESPEasy/WebServer.ino line 5798
0x4024a8ef: handle_schedule() at /home/john/Arduino/scetchbooks/ESPEasy/WebServer.ino line 5798
0x4024a9f1: loop at /home/john/Arduino/scetchbooks/ESPEasy/WebServer.ino line 5798
0x40258b69: esp_schedule at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 95
0x40258bd8: loop_wrapper at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 125
0x40100739: cont_wrapper at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/cont.S line 81

```

Serial log shows: 
```

88 chunk size:400
35365 : sendcontent free: 14688 chunk size:400
35394 : sendcontent free: 14552 chunk size:400
35430 : sendRuleDebug: 111: 
content free: 16008 chunk size:400
35435 : sendcontent free: 14664 chunk size:400
35451 : Rules : Save rulesSet: 1 currentSet:RuleDebug: 000: endon
 1 rules.length(): 458 Write to file: rules1.txt
35452 :  Webserver args: 0: 'set' length: 1 1: 'rules' leRuleDebug: 001: 
ngth: 458
35456 : sendcontent free: 14656 chunk size:400
35528 : sendcontent free: 15264 chunk size:400
3560RuleDebug: 100: relayoff do
5 : sendcontent free: 15224 chunk size:400
35640 : sendcontent free: 16048 chunk size:320
35644 : RuleDebug: 101: gpio,12,0
sendcontent free: 14704 chunk size:0
35770 : sendcontent free: 16672 chunk size:344
35774 : sendconteRuleDebug: 101: gpio,0,1
nt free: 16672 chunk size:0
35780 : ACT  : gpio,on System#Boot do
35783 : SW   : GPIO 0 Set to 0
358RuleDebug: 101: 
01 : ACT  : event,toggle
35857 : ACT  : gpio,12,1
35860 : SW   : GPIO 12 Set to 1
35876 : ACT  : gpio,0,0
3RuleDebug: 000: endon
5878 : SW   : GPIO 0 Set to 0
35894 : ACT  : 
RuleDebug: 100: toggle do
RuleDebug: 101: if [relaystate#relaystate]=1
RuleDebug: 101: event,relayoff
RuleDebug: 101: else
RuleDebug: 101: event,relayon
RuleDebug: 101: endif
RuleDebug: 000: endon
RuleDebug: 100: button#state=1 do
RuleDebug: 101: event,toggle
RuleDebug: 000: endon
RuleDebug: 100: mqttimport#mqttoh_relaycmd=1 do
RuleDebug: 101: event,relayon
RuleDebug: 000: endon
RuleDebug: 100: mqttimport#mqttoh_relaycmd=0 do
RuleDebug: 101: event,relayoff
RuleDebug: 000: endon

Soft WDT reset

```

s0170071, 2018-09-28:
possible fix (crash gone when you save the rules and memory drain gone too: 

place a yield() in handle_schedule().

```
void handle_schedule() {
  unsigned long timer;
  const unsigned long mixed_id = msecTimerHandler.getNextId(timer);
  if (mixed_id == 0) {
    // No id ready to run right now.
    // Events are not that important to run immediately.
    // Make sure normal scheduled jobs run at higher priority.
    backgroundtasks();
    process_system_event_queue();
    return;
  }
  const unsigned long timerType = (mixed_id >> TIMER_ID_SHIFT);
  const unsigned long mask = (1 << TIMER_ID_SHIFT) -1;
  const unsigned long id = mixed_id & mask;
**yield();**
  switch (timerType) {
    case CONST_INTERVAL_TIMER:
      process_interval_timer(id, timer);
      break;
    case PLUGIN_TASK_TIMER:
      process_plugin_task_timer(id);
      break;
    case TASK_DEVICE_TIMER:
      process_task_device_timer(id, timer);
      break;
  }
}

```

then the serial log shows when you push save:

```
166977 : Rules : Save rulesSet: 1 currentSet: 1 rules.length(): 460 Write to file: rules1.txt
166978 :  Webserver args: 0: 'set' length: 1 1: 'rules' length: 460
166982 : sendcontent free: 16048 chunk size:400
167052 : sendcontent free: 15688 chunk size:400
167128 : sendcontent free: 16320 chunk size:400
167161 : sendcontent free: 15128 chunkRuleDebug Processing:rules1.txt
     flags CMI  parse output:
RuleDebug: 100: system#boot do
 size:322
167166 : sendcontent free: 15128 chunk size:0
167287 : sendcontent free: 16272 RuleDebug: 101: 
chunk size:400
167293 : sendcontent free: 15600 chunk size:400
167299 : sendcontent free: 15144 chunk size:350RuleDebug: 000: endon

167304 : sendcontent free: 14624 chunk size:0
167344 : sendcontent free: 16576 chunk size:118
167346 :RuleDebug: 110: relayon do
 sendcontent free: 15904 chunk size:0
167462 : Command: event
167463 : EVENT: RelayOn
RuleDebug: 111: gpio,12,1
167523 : ACT  : gpio,12,1
167525 : SW   : GPIO 12 Set to 1
RuleDebug: 111: gpio,0,0
167535 : ACT  : gpio,0,0
167538 : SW   : GPIO 0 Set to 0
RuleDebug: 111: 
167547 : ACT  : 
RuleDebug: 000: endon
RuleDebug: 001: 
RuleDebug: 100: relayoff do
RuleDebug: 101: gpio,12,0
RuleDebug: 101: gpio,0,1
RuleDebug: 000: endon
RuleDebug: 100: toggle do
RuleDebug: 101: if [relaystate#relaystate]=1
RuleDebug: 101: event,relayoff
RuleDebug: 101: else
RuleDebug: 101: event,relayon
RuleDebug: 101: endif
RuleDebug: 000: endon
RuleDebug: 100: button#state=1 do
RuleDebug: 101: event,toggle
RuleDebug: 000: endon
RuleDebug: 100: mqttimport#mqttoh_relaycmd=1 do
RuleDebug: 101: event,relayon
RuleDebug: 000: endon
RuleDebug: 100: mqttimport#mqttoh_relaycmd=0 do
RuleDebug: 101: event,relayoff
RuleDebug: 000: endon
167612 : EVENT: RelayOn Processing time:150 milliSeconds
RuleDebug Processing:rules1.txt
     flags CMI  parse output:
RuleDebug: 100: system#boot do
167639 : Command: event
167640 : EVENT: toggle
RuleDebug: 101: 
RuleDebug: 000: endon
RuleDebug: 100: relayon do
RuleDebug: 101: gpio,12,1
RuleDebug: 101: gpio,0,0
RuleDebug: 101: 
RuleDebug: 000: endon
RuleDebug: 001: 
RuleDebug: 100: relayoff do
RuleDebug: 101: gpio,12,0
RuleDebug: 101: gpio,0,1
RuleDebug: 000: endon
RuleDebug: 110: toggle do
RuleDebug: 111: if 0=1
167686 : Read settings: TaskSettings index: 0
167689 : Read settings: TaskSettings index: 1
167692 : DEBUG DEV: ParseRuleDebug: 111: event,relayoff
d String='if 0'
167693 : Read settings: TaskSettings index: 3
167697 : Lev.1: [if 0=1]=false
RuleDebug: 111: else
167725 : Lev.1: [else]=true
RuleDebug: 111: event,relayon
167732 : ACT  : event,RelayOn
RuleDebug: 111: endif
RuleDebug: 000: endon
RuleDebug: 100: button#state=1 do
RuleDebug: 101: event,toggle
RuleDebug: 000: endon
RuleDebug: 100: mqttimport#mqttoh_relaycmd=1 do
RuleDebug: 101: event,relayon
RuleDebug: 000: endon
RuleDebug: 100: mqttimport#mqttoh_relaycmd=0 do
RuleDebug: 101: event,relayoff
RuleDebug: 000: endon
167774 : EVENT: toggle Processing time:134 milliSeconds
167797 : Command: 
167797 : Command unknown: ""
RuleDebug Processing:rules1.txt
     flags CMI  parse output:
RuleDebug: 100: system#boot do
167804 : Command: event
167805 : EVENT: RelayOn
RuleDebug: 101: 
RuleDebug: 000: endon
RuleDebug: 110: relayon do
RuleDebug: 111: gpio,12,1
167829 : ACT  : gpio,12,1
167831 : SW   : GPIO 12 Set to 1
RuleDebug: 111: gpio,0,0
167841 : ACT  : gpio,0,0
167843 : SW   : GPIO 0 Set to 0
RuleDebug: 111: 
167852 : ACT  : 
RuleDebug: 000: endon
RuleDebug: 001: 
RuleDebug: 100: relayoff do
RuleDebug: 101: gpio,12,0
RuleDebug: 101: gpio,0,1
RuleDebug: 000: endon
RuleDebug: 100: toggle do
RuleDebug: 101: if [relaystate#relaystate]=1
RuleDebug: 101: event,relayoff
RuleDebug: 101: else
RuleDebug: 101: event,relayon
RuleDebug: 101: endif
RuleDebug: 000: endon
RuleDebug: 100: button#state=1 do
RuleDebug: 101: event,toggle
RuleDebug: 000: endon
RuleDebug: 100: mqttimport#mqttoh_relaycmd=1 do
RuleDebug: 101: event,relayon
RuleDebug: 000: endon
RuleDebug: 100: mqttimport#mqttoh_relaycmd=0 do
RuleDebug: 101: event,relayoff
RuleDebug: 000: endon
167917 : EVENT: RelayOn Processing time:112 milliSeconds
167945 : Command: 
167945 : Command unknown: ""
168396 : sendcontent free: 18544 chunk size:400
168402 : sendcontent free: 17872 chunk size:400
168407 : sendcontent free: 17416 chunk size:213
168434 : sendcontent free: 18760 chunk size:0
168471 : sendcontent free: 18760 chunk size:118
168473 : sendcontent free: 18088 chunk size:0
169517 : sendcontent free: 18760 chunk size:118
169519 : sendcontent free: 18088 chunk size:0
169608 : sendcontent free: 18760 chunk size:118
169612 : sendcontent free: 18240 chunk size:0
170592 : sendcontent free: 18760 chunk size:118
170595 : sendcontent free: 18760 chunk size:0
170661 : sendcontent free: 18760 chunk size:118
170665 : sendcontent free: 18760 chunk size:0
171052 : UDP  : EC:FA:BC:87:76:B3,192.168.1.61,3

```

s0170071, 2018-10-04:
I just tried the latest build. My self-triggering rules now do drain the free mempory very  quickly. The ESP is out of memory after about 7 seconds and crashes. That used to work better. 

Tested with mega 20181004 and latest core, also did add -DNO_EXTRA_4K_HEAP -DPLUGIN_BUILD_DEV to platform.txt.

TD-er, 2018-10-04:
As far as I know, the ore 2.4.1 does nothing with `NO_EXTRA_4K_HEAP`.

And what rules do deplete the memory so fast? Can you post them here?

s0170071, 2018-10-04:
the rules are in the first post of this issue. They are rather stupid but I consider them an excellent test for stability.

giig1967g, 2018-10-04:
Hi @s0170071:

can you try avoid using the same name for device and task:
`if [RelayState#RelayState]=1`

I am not sure if this generates a sort of loop...

Just try and see if something changes:
`if [Relay#State]=1`

TD-er, 2018-10-04:
Maybe this is also related to the commands still using the queue?
I really should look into that, since the test PR I made still causes reboots apparently.
And I really don't see why, since it is the old behavior.

s0170071, 2018-10-04:
@giig1967g I know that. As we are constantly struggling with stability I did deliberately search for a setup that troubles the ESP faster and more reliable :-) Just for testing. And besides that, the rules I chose "should" work.

TD-er, 2018-10-04:
I agree it may be triggering some sort of 'stack overflow'.
But I think it should not be allowed to do that. So this is indeed a proper test to trigger that issue, since other 'real world' scenarios may be very hard to reproduce.

TD-er, 2018-10-23:
The log in this topic mentions MQTTimport.
Yesterday I added a delay(1) to the readByte part of MQTT client PubSubClient.
Can you please test if this is now still an issue?

Apart from that, a number of other fixes have been done lately which may be related to this issue.

TD-er, 2018-12-02:
I guess with today's commits, this one could be fixed.
If it is indeed fixed, please close it.

closed 2019-10-10 by TD-er