letscontrolit/ESPEasy #1820: Request: Log filter

Domosapiens, 2018-09-28:
Ref: mega-20180923

### Summarize of the feature request
Please offer the possibility to filter the Log on a keyword with wildcards

### Requested behavior
Adaptation of the Tools/Log Web GUI with a keyword field.

With a filter, like:
  `Error`
Log shows only occurrences like:
DS : Temperature: Error! (28-da-30-1-0-0-80-92)
DS : Temperature: Error! (28-ff-7c-68-a2-16-3-95)

With a filter, like:
  `Error*28-ff`
Log shows only occurrences like:
DS : Temperature: Error! (28-ff-7c-68-a2-16-3-95)

Also applicable to show re-boot's, WIFI re-connects etc.

### Actual behavior
Log is now overloaded with non-specific info in cases of faultfinding.

Domosapiens, 2018-09-28:
@Grovkillen : I would prefer a short term solution for bug finding in latest releases to support a new stable release.
Since ESPEasy_v2.0.0-dev11, I see a huge performance improvement but also a stability decrease (watchdog reboots, much more DS18B20 read errors).
 I'm afraid that a new GUI will introduce a new period of instability.
First a stable and then renewal?

Grovkillen, 2018-09-28:
I hear ya. I will look into a intermediate solution.

workgroupengineering, 2018-09-28:
I have the same problem too. I'm working on the PR # 1830 and I need to monitor the efficiency. I weigh that it is necessary to make a filter for scope.

Example:

```c
addLog (LOG_LEVEL_ERROR, F ("Error:") /*  Message */ , F ("Rules"), F ("Core"), ... / * Scopes * /);
```
So that you can filter in addition to the log level also for the scope.

TD-er, 2018-09-28:
Would it also help to add a command to filter only logs sent by a specific plugin/command?

Grovkillen, 2018-09-29:
Yeah that would be nice well @TD-er

Domosapiens, 2018-09-29:
@TD-er : Yep ... would be nice.
Also nice ... the calling module before manual reboot ;)
But may I ask for KISS .... just start with a simple GUI Log filter ??

TD-er, 2018-09-29:
@Domosapiens Both require adding extra information to the log indicating its source.
Or you mean only showing the current levels? (not up-to like it is now)

s0170071, 2018-09-29:
Does anyone of you know the openhab log viewer ? 
They filter the log on-the-fly in the browser. Watch it here: 
https://www.youtube.com/watch?v=5R9Oc8xINGU
at time code 10:50 min.

TD-er, 2018-09-29:
I was thinking about including the required log level in the http get request.
Filtering client side still means the log buffer may fill up and it takes resources creating the logs.

Domosapiens, 2018-09-29:
> I was thinking about including the required log level in the http get request

Seems to me the optimal solution

> Or you mean only showing the current levels?

A quick and dirty in the browser solution?

Up to you @TD-er. 
I only can support testing towards the stable release.

workgroupengineering, 2018-10-02:
@TD-er I working to scope log in this [branch](https://github.com/workgroupengineering/ESPEasy/tree/feature/scoped_log), but I have a doubt.

When loglevelActiveFor is invoked with just logLevel and there are log scopes in the settings, how should it behave loglevelActiveFor ?

TD-er, 2018-10-02:
@workgroupengineering At this moment the check is a "less than or equal" like test.
So if we allow to set each level on/off, the levels should be considered as a bitmap. Then the compare should be an AND with only that bit set as mask.

Grovkillen, 2020-01-09:
The new GUI now have filtering...
[image]

tonhuisman, 2023-09-02:
Not sure how relevant this still is, can possibly be closed.

closed 2023-09-02 by TD-er