letscontrolit/ESPEasy #1822: Web-update on Lolin D1 Mini Pro fails

Heiend, 2018-09-29:
I compiled ESPEasy (20180923) for the Wemos LOLIN D1 Mini Pro with 16 MByte Flash and the device runs without problems except the update via web (Tools/Firmware/Load)
> Update error: ERROR[4]: Not Enough Space
I was successfully updating a 4MByte ESP8266-board via web update.
There ist also a problem with the MD5-check on the LOLIN D1 Mini Pro but I don't think that causes the update-fail:
> Build Md5 | 4d44355f4d44355f4d44355f4d44355f
> Md5 check | fail !

TD-er, 2018-09-29:
The MD5 check fails when doing your own builds, without calling the python script to update the checksum.
How is your flash partitioned? (sketch size, free size after sketch and size of SPIFFS)

Heiend, 2018-09-29:
In Arduino I have only one option for the flash size: 16M (15M SPIFFS).

If the failing MD5 check doesnt cause any problems I would look for that later :-)

TD-er, 2018-09-29:
The layout is as follows:

- Current active sketch
- Space for OTA
- SPIFFS

If the SPIFFS is 15MB, this means there is 1MB left for sketch + OTA.
The current sketch is over 512 kB, so it is not possible with that layout to perform OTA updates.

Heiend, 2018-09-29:
Ah, ok. Is it possible to define the layout by myself (in the conf file of the board) or it is defined by factory?

TD-er, 2018-09-29:
I am not sure about the ESP8266. For the ESP32 it is very easy to do, but for ESP8266 it is included in the packages you download with the platformIO packages.
They recently added a special layout on my request for 2 MB flash units (Shelly 1 modules) specific for OTA like you experience now.

If you don't need the extra space at this moment, you can choose the 4MB builds, which have enough space reserved for OTA.
I will try to see how we can create some layout with 14 MB SPIFFS and thus 2 MB for sketch + OTA

TD-er, 2018-09-29:
It seems they also added a 14M/16M SPIFFS layout when also adding the requested special layout for 2 MB flash.
See: https://github.com/esp8266/Arduino/pull/5103/files
https://github.com/esp8266/Arduino/pull/5103/files#diff-aa150fc20266cc52096e8224a8343ac3
This is only available in the staging area at this moment, so in due time I will add it to the main builds.

TD-er, 2019-10-27:
Just as a remark (probably to myself for later) about the status of this issue.

There is a build option for 16M flash layout, with 14 MB SPIFFS.
The problem is that large SPIFFS partitions are unworkable slow.

We may need to test with LittleFS as file system.

closed 2023-09-03 by TD-er