letscontrolit/ESPEasy #1828: Rules: order of code execution

kluszczyn, 2018-09-30:
### Summarize of the problem/feature request

I have esp node with PIR and LDR (and SI + button - not relevant to issue).
When dark and motion detected it should switch on led strips.

With releases up to [20180804](https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20180804) this setup was working as expected, with [20180808](https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20180808) and following is not working. 
**Here is a devices page**
[image]

**Rules**
When motion detected, rule checks value X of dummy device Var. X keeps level of light or 0 if led strips were switched off by node (either by timer or manually via button).
If X=0 (means led strip is off), then node should measure light level, store it in X var. Next rule should compare X with light threshold (380 in this case) and if above, turn on led + set timer1 or set X to 0 otherwise.
This logic allows to keep light on for consequential motion triggers when led strip is on and therefore measured light is below threshold value (i have here inverted logic due to LDR values - low values means high level of light).  

Note: all comments in below code was added for reporting issue purpose, I do not have them in production env.
```
On Clock#Time do
  Publish easy/209/status/sys,{"uptime":%uptime%,"rssi":%rssi%,"wifi_ch":%wi_ch%,"time":%unixtime%,"ip":%ip%}
endon

//main logic for node
On PIR#Switch=1 do
  TaskValueSet 5,2,[Var#C]+1 //increase counter, stat purposes
  if [Var#X]=0 //if X=0 means led strip is off
    TaskRun,4 //measure light level LDR
    TaskValueSet 5,1,[LDR#Analog] //store measured level in Var#X
  endif
  if [Var#X]>380 //light level threshold, when>380 switch on led strip 
    PWM,12,1000,2500
    timerSet,1,60 //set (re-set) timer to switch off led strip after 60s
  else
    TaskValueSet 5,1,0.00
  endif  
endon

On Rules#Timer=1 do
  PWM,12,0,1000
  TaskValueSet 5,1,0 //set Var#X to 0
  TaskValueSet 5,3,[Var#L]+1
endon

//switching manually on/off led strip
On Button#Switch=1 do
  timerSet,2,1
  PWM,12,1000,300
  timerSet,1,180
endon

On Rules#Timer=2 do
  if [Button#Switch]=1
   PWM,12,0,100
   TaskValueSet 5,4,[Var#M]+1
   TaskValueSet 5,1,0
  endif
endon
```

**Logs**
```
114077: EVENT: Clock#Time=Sun,06:38
114089: ACT : Publish easy/209/status/sys,{'uptime':2,'rssi':-72,'wifi_ch':1,'time':1538282280,'ip':192.168.1.209}
114124: EVENT: Clock#Time=Sun,06:38 Processing time:48 milliSeconds
114132: Command: publish
114754: ACT : TaskValueSet 5,2,0+1
114782: [if 0.00=0]=true
114785: ACT : TaskRun,4
114811: ACT : TaskValueSet 5,1,964.09
114842: [if 0.00>380]=false
114846: else = true
114849: ACT : TaskValueSet 5,1,0.00
114882: EVENT: PIR#Switch=1.00 Processing time:154 milliSeconds
114906: MQTT : easy/209/PIR/Switch 1
114918: Command: taskvalueset
114923: Command: taskrun
114929: ADC : Analog value: 961.703
114932: EVENT: LDR#Analog=961.70
114960: EVENT: LDR#Analog=961.70 Processing time:28 milliSeconds
115931: MQTT : easy/209/LDR/Analog 961.70
115944: Command: taskvalueset
115947: Command: taskvalueset
117437: SW : Switch state 0 Output value 0
117443: EVENT: PIR#Switch=0.00
117467: EVENT: PIR#Switch=0.00 Processing time:25 milliSeconds
117490: MQTT : easy/209/PIR/Switch 0
123966: ACT : TaskValueSet 5,2,1+1
123994: [if 0.00=0]=true
123997: ACT : TaskRun,4
124024: ACT : TaskValueSet 5,1,961.70
124054: [if 0.00>380]=false
124058: else = true
124061: ACT : TaskValueSet 5,1,0.00
124094: EVENT: PIR#Switch=1.00 Processing time:154 milliSeconds
124118: MQTT : easy/209/PIR/Switch 1
124132: Command: taskvalueset
124135: Command: taskrun
124140: ADC : Analog value: 964.148
124144: EVENT: LDR#Analog=964.15
124172: EVENT: LDR#Analog=964.15 Processing time:29 milliSeconds
125143: MQTT : easy/209/LDR/Analog 964.15
125148: Command: taskvalueset
125163: Command: taskvalueset
126652: SW : Switch state 0 Output value 0
126658: EVENT: PIR#Switch=0.00
126682: EVENT: PIR#Switch=0.00 Processing time:24 milliSeconds
126705: MQTT : easy/209/PIR/Switch 0
132581: ACT : TaskValueSet 5,2,2+1
132609: [if 0.00=0]=true
132612: ACT : TaskRun,4
132639: ACT : TaskValueSet 5,1,964.15
132670: [if 0.00>380]=false
132675: else = true
132678: ACT : TaskValueSet 5,1,0.00
132709: EVENT: PIR#Switch=1.00 Processing time:154 milliSeconds
132733: MQTT : easy/209/PIR/Switch 1
132747: Command: taskvalueset
132749: Command: taskrun
132755: ADC : Analog value: 965.280
132759: EVENT: LDR#Analog=965.28
132786: EVENT: LDR#Analog=965.28 Processing time:28 milliSeconds
133758: MQTT : easy/209/LDR/Analog 965.28
133763: Command: taskvalueset
133775: Command: taskvalueset
135291: EVENT: Si#Temp=22.57
135314: EVENT: Si#Temp=22.57 Processing time:25 milliSeconds
135317: EVENT: Si#Hum=38.50
135343: EVENT: Si#Hum=38.50 Processing time:27 milliSeconds
135366: MQTT : easy/209/Si/Temp 22.57
135367: MQTT : easy/209/Si/Hum 38.50
135376: SW : Switch state 0 Output value 0
135382: EVENT: PIR#Switch=0.00
135407: EVENT: PIR#Switch=0.00 Processing time:26 milliSeconds
136392: MQTT : easy/209/PIR/Switch 0
136406: ADC : Analog value: 956.933
136409: EVENT: LDR#Analog=956.93
136435: EVENT: LDR#Analog=956.93 Processing time:26 milliSeconds
137417: MQTT : easy/209/LDR/Analog 956.93
137425: Dummy: value 1: 0.00
137426: Dummy: value 2: 3.00
137427: Dummy: value 3: 0.00
137428: Dummy: value 4: 0.00
137431: EVENT: Var#X=0.00
```

It looks like second ``if [Var#X]>380`` logic is executed before LDR takes reading and stores them in Var X:

```
On PIR#Switch=1 do
  TaskValueSet 5,2,[Var#C]+1
  if [Var#X]=0
    TaskRun,4
    TaskValueSet 5,1,[LDR#Analog]
  endif
  if [Var#X]>380
    PWM,12,1000,2500
    timerSet,1,60
  else
    TaskValueSet 5,1,0.00
  endif  
endon
```

```
114782: [if 0.00=0]=true
114785: ACT : TaskRun,4
114811: ACT : TaskValueSet 5,1,964.09
114842: [if 0.00>380]=false
```

As a result led strip is never turned on when motion detected. 

Release ESP_Easy_mega-20180804_normal_ESP8266_4096.bin is last one working correctly with above setup and rules. All new ones, including last one, build ESP_Easy_mega-20180930_normal_ESP8266_4096.bin, do not work correctly.

There was some changes in rule handling in 20180808:
```
      [Scheduler] Rename some functions and clean-up
      [Notifications] Schedule NPlugin_ptr calls async
      [MQTT] Schedule from callback   @@functions
      [Rules] Split parsing single line from rulesProcessingFile
      [Rules] Split processing matched rule in a separate function
      [Commands] Return String output per command
      [Schedule] Scheduled command timer
      [Schedule] Run command from webpage immediately
```

### System configuration

Hardware:
ESP8266 (ESP12)
Controller: MQTT (openhab)

ESP Easy version: >= 20180808. Tested with ESP_Easy_mega-20180XXX_normal_ESP8266_4096.bin

TD-er, 2018-09-30:
My first guess is that this is one of those issues related to running commands from the queue instead of executing them immediately. (last 2 items of your 20180808 changes list)
I tried to revert that change about 2 weeks ago, but apparently that resulted in boot loops.
See:
- https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20180915
- https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20180916

On a side note, may I compliment you with the extensive research you did for this issue?
If all reported issues were this clearly made, it would save a lot of time getting all information.

kluszczyn, 2018-09-30:
Thanks @TD-er for kind words. At least this is -trying reporting issues accurately - how I can contribute to this great project.

For time being I will revert to release 20180804, but as I have two nodes with the same rules logic related to LDR part, I can test upon request new code, provide logs, etc.

Grovkillen, 2018-09-30:
Officially we do not support multiple if's in the same on. But as you stated it did somewhat work. We need to address that more thoroughly since we're now will start using the queued approach.

kluszczyn, 2018-09-30:
@Grovkillen, I was not aware that multiply if's is not supported in one on-endon block; it was clear to me, that nested if's are not permitted. Good to know. As you said, it was working somehow for me with old fw.

1. I've modified rule in below way to remove multiply if's:
```
On PIR#Switch=1 do //motion detected
  TaskValueSet 5,2,[Var#C]+1 //counter, for stats only
  if [Var#X]=0 // 0 means led strip is off, let's check light level
    TaskRun,4 //trigger LDR to measure light level
    TaskValueSet 5,1,1 //set the flag motion detected + led strip is off
  else
    timerSet,1,60 //if flag is set to 1 means that led strip is on and we just need to reschedule timer to keep led strip on
  endif
endon

On Var#X=1 do //when flag set to 1 - see above desc
  if [LDR#Analog]>380 //if light level above threshold switch led strip on
    PWM,12,1000,2500 //switch led strip on
    timerSet,1,60 //set timer1 for 60s
  endif 
endon 
```
Logic is the same as from original code. Var#X is now the flag set to 1 when motion detected and light level > threshold. Subsequent motions triggers only timer reschedule (keeps led strip on) without measuring light. 

2. Then I updated node to latest build (ESP_Easy_mega-20180930_normal_ESP8266_4096.bin).

It works as expected, but with one exception: led strip is switched on not just after motion detected, but with delay. I observed, that led strip is switched on when event ``EVENT: LDR#Analog`` is fired, but one from device 'schedule', not from ``TaskRun`` command. So in worst scenario delay in switching on led strips can be nearly 60s, as LDR is sampled every 60s (in my case). 

**Log below**
-Motion is detected: 4077.391s
-LDR event (from TuskRun): 4077.471s
-LDR| event (from device 'schedule'): 4095.305s
-GPIO set to on: 4098.946s

Led strip is switched on with ~21s delay

```
4077391: EVENT: PIR#Switch=1.00
4077414: ACT : TaskValueSet 5,2,98+1
4077427: Command: taskvalueset
4077446: [if 0.00=0]=true
4077449: ACT : TaskRun,4
4077462: Command: taskrun
4077468: ADC : Analog value: 940.231
4077471: EVENT: LDR#Analog=940.23
4077500: EVENT: LDR#Analog=940.23 Processing time:28 milliSeconds
4077522: MQTT : easy/209/LDR/Analog 940.23
4077528: ACT : TaskValueSet 5,1,1
4077542: Command: taskvalueset
4077546: else = false
4077567: EVENT: PIR#Switch=1.00 Processing time:177 milliSeconds
4078549: MQTT : easy/209/PIR/Switch 1
4079963: SW : Switch state 0 Output value 0
4079966: EVENT: PIR#Switch=0.00
4079993: EVENT: PIR#Switch=0.00 Processing time:28 milliSeconds
4080017: MQTT : easy/209/PIR/Switch 0
4095200: EVENT: Si#Temp=23.05
4095227: EVENT: Si#Temp=23.05 Processing time:29 milliSeconds
4095230: EVENT: Si#Hum=43.60
4095256: EVENT: Si#Hum=43.60 Processing time:28 milliSeconds
4095280: MQTT : easy/209/Si/Temp 23.05
4095281: MQTT : easy/209/Si/Hum 43.60
4095302: ADC : Analog value: 1000.054
4095305: EVENT: LDR#Analog=1000.05
4095332: EVENT: LDR#Analog=1000.05 Processing time:27 milliSeconds
4098946: SW : GPIO 12 Set PWM to 1000
4098951: ACT : timerSet,1,60
```
So this behavior has something with queued commands and what triggers commands execution from queue.

Hope this helps to trace root cause of issue.

TD-er, 2018-09-30:
Yep, the commands are currently placed in a queue, which is processed with lower priority.
So I will disable the command queue, due to the unforeseen effects in rules handling.

Last time I did that, there were reports of crashes, so I have to look into that before I merge such changes again.
Maybe it is as simple as not checking for an empty queue.

TD-er, 2018-11-27:
This should be fixed now with the next nightly build.
I set it to fixed now. Please close when confirmed fixed.

Grovkillen, 2018-12-14:
I close this.

closed 2018-12-14 by Grovkillen