letscontrolit/ESPEasy #1830: Anyone has same hardware and can proof my story ?

Sasch600xt, 2018-09-30:
Hello,

i am close to give up on here. I would love to work with ESP-EASY MEGA, but somehow i feel like noone is beliving me anymore.

Does anyone have a Nodemcu Lolin V3 board and can proof for example System Uptime plugin and setting up an Domoticz HTTP controller ?
This is not working for me since at least 10 releases.
I stuck at 20180826.

I did try an workaround to make sure it is not the fault of my PHP receiver script and did set up an "real" Domoticz server for testing. Also there i do not receive messages with a higher release as 20180826.

So questions are:

- am i too stupid ?
- is there a bug in EspEasy which noone detected so far ?
- is using Domoticz HTTP controller possible for others with same hardware board like mine over here ?

Thanks for your help
Sascha

TD-er, 2018-09-30:
Can you post screenshots of your settings?
I don't believe the hardware is an issue here, especially since it has been working in the past.

Sasch600xt, 2018-09-30:
Hello TD-er......good morning :)

here again my screenshots (older version, but same issue)

Thank you for your help
[issue_sascha.zip](https://github.com/letscontrolit/ESPEasy/files/2431357/issue_sascha.zip)

ShardanX, 2018-09-30:
I agree with TD-er: Most likely not a hardware problem.
I've running several nodeMCUs (LoLin and others) with newer versions then 20180826 for testing.
Not using domoticz so I can't say more.

sakinit, 2018-09-30:
[image]

Using test setup with Nodemcu V3 board. Your 2018-09-12 12.46.01 shows an up-time of 0 0 0 and a Load of 100%, maybe immediately after start-up?

Sasch600xt, 2018-09-30:
yes......if you disable the controller and after a reboot, the Load should be normal and the uptime counter counts again.

Sasch600xt, 2018-09-30:
or you could try to send an http command from commandline in tools. this chrashed my nodemcu

sakinit, 2018-09-30:
e.g. sysload and lowmem are working fine

Sasch600xt, 2018-10-01:
i did try the new image 20181001 right now.
From the moment  i set up the domoticz controller, uptime in main page freezed.
I did not set up a plugin at all so far. Just setting up the controller was enough to freeze at least uptime.

I hope this helps to find the bug

Have a great evening
Sascha

sakinit, 2018-10-01:
Also after a factory reset?

Sasch600xt, 2018-10-02:
at the moment i always start from scratch (4Mb blank image first)

Sasch600xt, 2018-10-02:
is it a problem because i do NOT have a Domoticz server running here ? I use my own receiver script for PHP. So could you please try to set an Domoticz Controller in ESPEasy while NO Domoticz server is running and tell me the behavior ? Thank you

TD-er, 2018-10-02:
So, I'm now planning to look into this one, since I finished all other issues I needed to get out of my head :)

Sasch600xt, 2018-10-02:
ok......thank you......gimme a shout if i can help you somehow

TD-er, 2018-10-02:
This is still the same issue as #1729 right?
See this comment:
https://github.com/letscontrolit/ESPEasy/issues/1729#issuecomment-425578074

> If i use commandline in "tool" and send
> `SendToHTTP 192.168.123.10,8080,/json.htmtype=command&param=switchlight&idx=17&switchcmd=On`
> 
> i can receive once the message. After this short moment of glory the esp freezes and i have to do an hard reset. I can repeat this and the behavior is always the same.

Can you see (after the reboot) what the reboot cause was when you had this crash?

Sasch600xt, 2018-10-02:
how can i test this ?

TD-er, 2018-10-02:
It is the command you wrote in the other issue.
So I assume it has the right parameters in it and you can run it from the web interface: Tools => command input
[image]

And press Submit.
If the node reboots, you can go to the sysinfo page (main page => More Info) and look at the given Reset Reason.

Sasch600xt, 2018-10-02:
Yes...... i sent "SendToHTTP www.google.com,80,/search?q=espeasy" and i got an OK and no crash

TD-er, 2018-10-02:
So it looks like the host you're sending to is not replying to the HTTP request.
It is just accepting it and doesn't report back any acknowledgement.

TD-er, 2018-10-02:
In the `send_via_http` function, there is this code, which looks a bit strange to me:

```c++
  unsigned long timer = millis() + 200;
  while (!client.available() && !timeOutReached(timer))
    delay(1);

  // Read all the lines of the reply from server and print them to Serial
  while (client_available(client) && !success) {
```
Why not `return false;` when timeout is reached?
I will change this and make a test build for you to test it.

What build do you need? "normal 4096" I assume?

Sasch600xt, 2018-10-02:
yes....normal 4096.
And you are right.....i do not answere to the messages. i just read them out in my receiver script.
So it looks like in 20180826 there was no need to response to the request.

TD-er, 2018-10-02:
Around then I moved a lot of duplicate code to a single function.
Apparently there were subtle differences in handling for various situations where it is being used and this is one reproducible situation where it always tends to fail, where others may have crashes too (e.g. in controllers communicating via HTTP) but were unable to pinpoint where things crash.

I looked a bit further into the code and it looks like the WiFiClient is always using the same (default) timeout of 1000 msec. 
This is too long to keep the wifi connection alive, so I guess that's what may cause all other kinds of undefined behavior.

Sasch600xt, 2018-10-02:
i see.......so you opened the box of pandoras ? :)

TD-er, 2018-10-02:
I hope it is **THE** box we've been looking for and not just another one of her boxes.
It is building now, I will test it first on my node.
I expect to have a build ready in about 10 - 15 minutes from now.

Sasch600xt, 2018-10-02:
no stress, i was waiting weeks now :)

Thank you so much for taking the time to find the problem

TD-er, 2018-10-02:
Can you test [this test build](https://www.dropbox.com/s/qjzhid3pu7th2mb/ESPeasy_issue1830_set_timeout_normal_4096.zip?dl=0)  ?

Sasch600xt, 2018-10-02:
yes.....i tell you soon the result

TD-er, 2018-10-02:
That's great!
I will make a commit of it, so others can also test it.

TD-er, 2018-10-02:
I will close this one.
If it appears still not fixed, please re-open.

closed 2018-10-02 by TD-er