letscontrolit/ESPEasy #1832: _P035_IRTX.ino send Raw

kpalczewski, 2018-09-30:
I can't find a command to send RAW IR code for protocol that isn't supported by IRremoteESP8266 library.

I'm trying to send IR code using RAW option:
http://192.168.0.31/control?cmd=IRSEND,RAW,8550,4186,588,476,592,1524,592,1524,592,476,588,476,592,1524,594,474,588,1528,588,1528,592,476,588,476,588,1528,588,1528,592,476,590,1526,588,476,590,474,592,1524,592,476,588,1528,588,476,592,476,588,476,588,480,588,1524,592,476,588,1528,588,476,594,1522,592,1524,592,1528,588,1528,588

I have found RAW option here:
https://www.letscontrolit.com/wiki/index.php/ESPEasy_Command_Reference
I haven't found RAW option in code in irtx.ino plugin source code.
What URL Syntax is valid?
Is it possible to encode the RAW data in URL, so it would be possible to send longer IR codes?

What is URL syntax to send LG code?
I have found LG in source code here:
if (IrType.equalsIgnoreCase(F("LG"))) Plugin_035_irSender->sendLG(IrCode); 

but I haven't found info on how the URL has to be constructed.

uzi18, 2018-10-03:
syntax:
`http://192.168.0.31/control?cmd=IRSEND,RAW,base32hex_bits,frequency,pulse_length,blank_length`

kpalczewski, 2018-10-04:
thank you.

the ESPEasy shows only a handful of supported IR protocols. Apparently LG is also supported and not listed.
Syntax for LG:
http://192.168.0.31/control?cmd=IRSEND,LG,88C0051,28

closed 2018-10-04 by kpalczewski

uzi18, 2018-10-04:
@Grovkillen @TD-er wiki update?

uzi18, 2018-11-08:
or RAW to U32, and RAW2 to RAW?

jimmys01, 2018-11-08:
Raw2 is also a dase 32 typeish encoding