letscontrolit/ESPEasy #1835: ESP32 / Controller Generic HTTP / I2C : multiple values are not send by the controller

nicoguido, 2018-10-01:
### Summarize of the problem/feature request
I use version esp32test_1M8_partition from 20180930.
I use an I2C probe BME280 reporting temp, humidity and pressure. The 3 values are well reported in the device section of espeasy.
I user a controller Generic HTTP to send the 3 values to my domotic box.

With the ESP32, only temperature is send, but 11 times per cycle.
With the same configuration with ESP8266 (v2.0-20180322),  the 3 values (temp, humidity, pressure) are send, only 1 time per cycle.

here is the log of the domotic box for ESP32:
`We've got a request for /device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66 from 192.168.xxx.AAA
Calling Jeedom http://127.0.0.1:80/plugins/espeasy/core/api/jeeEspeasy.php?apikey=ZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZ&device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66&ip=192.168.xxx.AAA
Mon Oct 01 2018 11:52:11 GMT+0200 (CEST)Got response Value: 200
We've got a request for /device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66 from 192.168.xxx.AAA
Calling Jeedom http://127.0.0.1:80/plugins/espeasy/core/api/jeeEspeasy.php?apikey=ZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZ&device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66&ip=192.168.xxx.AAA
Mon Oct 01 2018 11:52:12 GMT+0200 (CEST)Got response Value: 200
We've got a request for /device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66 from 192.168.xxx.AAA
Calling Jeedom http://127.0.0.1:80/plugins/espeasy/core/api/jeeEspeasy.php?apikey=ZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZ&device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66&ip=192.168.xxx.AAA
Mon Oct 01 2018 11:52:12 GMT+0200 (CEST)Got response Value: 200
We've got a request for /device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66 from 192.168.xxx.AAA
Calling Jeedom http://127.0.0.1:80/plugins/espeasy/core/api/jeeEspeasy.php?apikey=ZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZ&device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66&ip=192.168.xxx.AAA
Mon Oct 01 2018 11:52:12 GMT+0200 (CEST)Got response Value: 200
We've got a request for /device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66 from 192.168.xxx.AAA
Calling Jeedom http://127.0.0.1:80/plugins/espeasy/core/api/jeeEspeasy.php?apikey=ZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZ&device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66&ip=192.168.xxx.AAA
Mon Oct 01 2018 11:52:12 GMT+0200 (CEST)Got response Value: 200
We've got a request for /device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66 from 192.168.xxx.AAA
Calling Jeedom http://127.0.0.1:80/plugins/espeasy/core/api/jeeEspeasy.php?apikey=ZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZ&device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66&ip=192.168.xxx.AAA
Mon Oct 01 2018 11:52:12 GMT+0200 (CEST)Got response Value: 200
We've got a request for /device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66 from 192.168.xxx.AAA
Calling Jeedom http://127.0.0.1:80/plugins/espeasy/core/api/jeeEspeasy.php?apikey=ZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZ&device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66&ip=192.168.xxx.AAA
Mon Oct 01 2018 11:52:13 GMT+0200 (CEST)Got response Value: 200
We've got a request for /device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66 from 192.168.xxx.AAA
Calling Jeedom http://127.0.0.1:80/plugins/espeasy/core/api/jeeEspeasy.php?apikey=ZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZ&device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66&ip=192.168.xxx.AAA
Mon Oct 01 2018 11:52:13 GMT+0200 (CEST)Got response Value: 200
We've got a request for /device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66 from 192.168.xxx.AAA
Calling Jeedom http://127.0.0.1:80/plugins/espeasy/core/api/jeeEspeasy.php?apikey=ZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZ&device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66&ip=192.168.xxx.AAA
Mon Oct 01 2018 11:52:13 GMT+0200 (CEST)Got response Value: 200
We've got a request for /device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66 from 192.168.xxx.AAA
Calling Jeedom http://127.0.0.1:80/plugins/espeasy/core/api/jeeEspeasy.php?apikey=ZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZ&device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66&ip=192.168.xxx.AAA
Mon Oct 01 2018 11:52:13 GMT+0200 (CEST)Got response Value: 200
We've got a request for /device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66 from 192.168.xxx.AAA
Calling Jeedom http://127.0.0.1:80/plugins/espeasy/core/api/jeeEspeasy.php?apikey=ZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZ&device=ESP32_Easy&taskid=0&cmd=Temperature&value=22.66&ip=192.168.xxx.AAA
Mon Oct 01 2018 11:52:13 GMT+0200 (CEST)Got response Value: 200
`

Here is the log for ESP8266:
`We've got a request for /device=ESP_Test&taskid=0&cmd=Temperature&value=21.6 from 192.168.xxx.BBB
Calling Jeedom http://127.0.0.1:80/plugins/espeasy/core/api/jeeEspeasy.php?apikey=ZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZ&device=ESP_Test&taskid=0&cmd=Temperature&value=21.6&ip=192.168.xxx.BBB
Mon Oct 01 2018 11:52:41 GMT+0200 (CEST)Got response Value: 200
We've got a request for /device=ESP_Test&taskid=0&cmd=Humidity&value=46.5 from 192.168.xxx.BBB
Calling Jeedom http://127.0.0.1:80/plugins/espeasy/core/api/jeeEspeasy.php?apikey=ZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZ&device=ESP_Test&taskid=0&cmd=Humidity&value=46.5&ip=192.168.xxx.BBB
Mon Oct 01 2018 11:52:42 GMT+0200 (CEST)Got response Value: 200
We've got a request for /device=ESP_Test&taskid=0&cmd=Pressure&value=1025.6 from 192.168.xxx.BBB
Calling Jeedom http://127.0.0.1:80/plugins/espeasy/core/api/jeeEspeasy.php?apikey=ZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZ&device=ESP_Test&taskid=0&cmd=Pressure&value=1025.6&ip=192.168.xxx.BBB
Mon Oct 01 2018 11:52:43 GMT+0200 (CEST)Got response Value: 200`

nicoguido, 2018-10-02:
Hi,
No change with mega-20181002

TD-er, 2018-10-07:
I can predict there is also no change in mega-20181008.
I simply haven't gotten to it yet.

nicoguido, 2018-10-13:
Hi,
I did some tests with mega versions for esp8266 and the change occurs between version ESP_Easy_mega-20180904_normal_ESP8266_4096.bin (working) and ESP_Easy_mega-20180908_normal_ESP8266_4096.bin (not working).
Hope it helps.

TD-er, 2018-10-20:
Should be fixed by PR #1926

nicoguido, 2018-10-21:
It is working on ESP8266.
Thx !

closed 2018-10-21 by nicoguido