letscontrolit/ESPEasy #1843: Test NaN in rules

artua, 2018-10-02:
### Summarize of the problem/feature request
Then we got NaN as a test condition, the level control and rules seems to not work. 

### Expected behavior

Somehow catch this: NaN should be tested as null or as inf.

### Actual behavior

Rules are not working

### Steps to reproduce

1. Connect DS18B20
2. Put a thermostat(LEVEL) with an input from TEMP
3. Disconnect DS10B20

The problem persists after powering off and on.

### System configuration

Hardware:
Sonoff Touch

ESP Easy version: Self compile, sorry

ESP Easy settings/screenshots: 
[image]
[image]
[image]
### Rules or log data

```
on RL2#SW=1 do
 if [TEMP2#TEMP]>1 or [TEMP2#TEMP]<40
  gpio,5,1
// it's ok
 else
  gpio,5,0
 endif
endon
```

TD-er, 2018-10-02:
In C++ any test with a NaN should fail, even when compared with itself.
In the rules we could add some check for == NaN.

For controllers, there is already a filter not to send values with NaN values.
I think you want to have the option to test for NaN in rules, but then you must have the option to actually test it ofcourse. :)

TD-er, 2018-10-02:
You may have a point here.
Maybe we should add an option per controller how to handle NaN values, since it depends on the controller if it can handle NaN values.

Domosapiens, 2018-10-02:
I observe with recent versions (tested from ESPEasy_mega-20180914 upto now) much more often a NaN value then with ESPEasy_v2.0.0-dev11.
With ESPEasy_v2.0.0-dev11 , I catches NAN on a display by coincidence, now I see NAN very often in the GUI.  
Is this a timing side-effect of the huge performance improvement?
Or a change in _P004_Dallas.ino since summer'17 ?

itProfi, 2018-10-03:
Welcome! - https://github.com/letscontrolit/ESPEasy/issues/1723  (((

TD-er, 2018-10-03:
I have a bunch of these sensors available now, and just moved it to the top of the "broken sensors" list in my Trello.

ArjenWols, 2019-11-28:
Hello Team,
When using battery powered ESPs I get a NaN when the battery power is too low.
For this I would like to test for the NaN on any device and when the equition is True I would like to sent an Alert to Domoticz.

Any idea how to attack this 'problem'?
Thanks for the help in advance.
Regards
Arjen

TD-er, 2019-11-28:
Hmm I don't think we have it yet.
Maybe you can compare the value with itself?

`if [task#var]<>[task#var]`
But I'm not sure how that will be processed in the code.
If the values are not converted to strings before being compared, then it should work, but I am afraid I have to change something in the code to make that work and maybe it is then better to have a function for it to check if a value is NaN, or like discussed before let a plugin send an event when an invalid value is being sent.

uzi18, 2019-11-28:
ok but battery level should not be NaN so it is something to fix, please show your task configuration for analog input

TD-er, 2019-11-28:
> I get a NaN when the battery power is too low.

@uzi18 I read it as: "some plugin gives NaN, when battery power is too low", not specifically the ADC plugin.

uzi18, 2019-11-28:
error "not read with only half open eye" :)