letscontrolit/ESPEasy #1844: Weblog works only a short time

DittelHome, 2018-10-03:
Using this firmware:
ESP_Easy_mega-20181003_normal_ESP8266_4096.bin
Hardware:
ESP8266 pro
Problem:
Under tools/advanded i set web log level to "debug dev". Syslog level and serial log level to none. 
Weblog works only for a short time, After i while (5-15 seconds) no more messages will be displayed.
If a refresh my browser tab (firefox), than i will get again messages for 5-15 seconds.
It doesnt matter, if autoscroll is set or not, same situation.
regards
Klaus

Grovkillen, 2018-10-03:
I will look into this and the filtering suggested.

DittelHome, 2018-10-04:
With todays release weblog is working as expected. (ESP_Easy_mega-20181004_normal_ESP8266_4096)
It seems that the core 2.4.1 is much more better...
Issue can be closed
best regards
Klaus

Grovkillen, 2018-10-04:
Ok, thanks for the feedback. We need to look into the matter for future updates of the core.