letscontrolit/ESPEasy #1847: 100 % CPU after sevreal hours ...

DittelHome, 2018-10-03:
Using this firmware:
ESP_Easy_mega-20181003_normal_ESP8266_4096.bin
Hardware:
ESP8266 pro
Devices:
Communication - Serial Server
Display - OLED SSD1306
Generic - MQTT Import

Problem:
After round about 10 hours i get 100% CPU load. After ESP restart cpu usage dropped to 25 %, as normal.
regards
Klaus

TD-er, 2018-10-03:
Do you also see other effects, like lost wifi (many reconnects), a sensor not responding or MQTT connection lost or reconnecting often?
Of is the node rebooting a lot (CPU load is 100% right after the start)

DittelHome, 2018-10-04:
The unit was unresponsive during high CPU-Load. And yes, during high load many reconnects encountered.
Also serial connection was dropped (timeout).
Today i have installed the release with core 2.4.1. It seems that all my problems are solved.

TD-er, 2018-10-04:
@DittelHome That last line sounds like music to my ears.
Only thing we now have to do is to get the memory usage down, reduce heap fragmentation and have a thorough look at the rules. Then we can call it stable :)

closed 2018-10-04 by TD-er