letscontrolit/ESPEasy #1850: "IP blocked" after entering "Client IP filtering" range 1.1.1.1 to 255.255.255.255

jayrisko, 2018-10-03:
### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

"IP blocked" on ESP8266/4MB after entering "Client IP filtering" values range 1.1.1.1 to 255.255.255.255

### Expected behavior

Should not block when specifying IP address range 1.1.1.1 to 255.255.255.255
### Actual behavior

"IP blocked" message - have to reflash the unit

### Steps to reproduce

1.  Go to the "Config" tab
2.  Under "Client IP filtering" section, select "Allow IP range"
3.  Enter 1.1.1.1 in "Access IP lower range"
4.  Enter 255.255.255.255 in "Access IP upper range"
5.  Click "Submit" button
6.  Go to "Tools" tab and click the "Reboot" button
7.  dohh...  get the "IP blocked" message and have to re-flash the unit...

### System configuration

Hardware:  Wemos D1 4M flash

ESP Easy version:   mega-20181002

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2018-10-03:
You can also disable the IP filtering using commands from the serial port.
This way you don't have to flash it again.
This disables the filtering until a reboot, but does allow you to re-enter the web config page.

Still it should work when using these settings, unless you appear to come from something like 1.1.1.0 :)

Budman1758, 2018-10-03:
Isn't it easier to just check "allow all"?  That works.

jayrisko, 2018-10-03:
Yes I could just "allow all" but there's still an apparent bug in the code with other possible side effects….  

I didn't realize I can disable the filtering via serial port. Could I also do that via an MQTT "cmd" string?

jayrisko, 2018-10-04:
The command "clearaccessblock" via the serial interface did allow me access over wifi to update the IP range filtering....  Thanks for your help with this tip!

-Jay

sanderlv, 2018-12-07:
Every version I put on my devices... always "IP blocked:" and then I have to go via serial and do clearaccessblock VERY ANNOYING :-)

TD-er, 2018-12-07:
What IP address do the ESP get and what IP address do you have when connecting to it?

sanderlv, 2018-12-07:
it gets 192.168.6.3/24 and I have 192.168.1.159/24

TD-er, 2018-12-07:
So the block is a valid block and not a bug.
How do you connect to it? Via a VPN or something like that?

Grovkillen, 2018-12-07:
@sanderlv how do you flash it? Using the provided ESP Easy Flasher or something else?

sanderlv, 2018-12-07:
In my home network. I have a router (my firewall and i have some subnets. 1 subnet is my office lan (192.168.1.159/24 is my PC and 1 subnet is my IOT lan, 192.168.6.3/24 is one of the easy devices... I would want that If I flash some it is NOT blocked by default if it is in the neighbour subnet... that way I always need to have a serial connection and physical access to devices before being able to configure them...

@ Grovkillen: I flash them via the download in the github releases. BTW, Nice tool in the latest trealse with the serial programmer included!!!

Might be an op tion to be able to "tap" clearaccessblock in there :-)

sanderlv, 2018-12-07:
@ TD-er: Normally (before the new flash application I used to do via http interface and my cellphone (wifi default etc. and choose right SSID), but now I use the new application provided (which is very cool!!!)

Grovkillen, 2018-12-07:
Thanks for the kudos! I could add that as an option yep

TD-er, 2018-12-07:
So adding more default settings in the flasher would help here?

sanderlv, 2018-12-07:
For Sure! I think lot of people use these devices remote or not in the same subnet as their controller/pc?

Grovkillen, 2018-12-07:
PR is live. #2142

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by TD-er