letscontrolit/ESPEasy #1854: DHT11 does not show dezimal values

Sasch600xt, 2018-10-04:
Hello all,
i am just wondering my DHT11 shows up the values only as Integer and not as expected as float.
it is always x.0 instead of x.>=0

i use the new image 20181004

here are two screenshots:
[image]

[image]

Thanks for your help
Sascha

TD-er, 2018-10-04:
I've looked at the source code and the datasheet.
This is what the datasheet has to say:

> Data consists of decimal and integral parts. A complete data transmission is 40bit, and the
> sensor sends higher data bit first.
> Data format: 8bit integral RH data + 8bit decimal RH data + 8bit integral T data + 8bit decimal T
> data + 8bit check sum. If the data transmission is right, the check-sum should be the last 8bit of
> "8bit integral RH data + 8bit decimal RH data + 8bit integral T data + 8bit decimal T data".

And this is the implementation in the source code:

```c++
      temperature = float(dht_dat[2]); // Temperature
      humidity = float(dht_dat[0]); // Humidity
```

In other words, only the integral part of the data appears to be used.
So it looks to me you found a bug, or at least some missing bits :)

I've looked at some code examples and found no library that would use the decimals.
Also the DHT11 has a quite large uncertainty range (+/- 2 degree), so I doubt is is really useful to show the decimals.

It seems the reported values are in [Q8.8](https://en.wikipedia.org/wiki/Q_(number_format)) format, so it is quite easy to add the decimals.

TD-er, 2018-10-04:
Can you test the PR #1855  ?

beicnet, 2018-10-04:
How I recall, DHT11 don't have any decimals only from DHT12 and above.

Sasch600xt, 2018-10-04:
i ordered a DHT22 and it will be there at Saturday.

@TD-er:
Do you want me to test your build with the DHT11 or should we wait for the DHT22 who has for sure float values ?

Thank you
Sascha

TD-er, 2018-10-04:
You can already test this build. The only change with the last nightly is the change for DHT11

Sasch600xt, 2018-10-04:
ok....but as i understood, there will be no float value as long i use the DHT11

TD-er, 2018-10-04:
The datasheet suggests there is a "decimal part" and I added something to interpret that data as I understood it from the datasheet

Sasch600xt, 2018-10-04:
ok....i will test now

TD-er, 2018-10-04:
Sorry, I now see I built the 1024 version.

Sasch600xt, 2018-10-04:
too late and my device crashed LOL :)

Sasch600xt, 2018-10-04:
right now i flash 4Mb blank

TD-er, 2018-10-04:
[Newer version](https://www.dropbox.com/s/qf7y558c1ll1vu9/issue1854_ESPeasy_normal_4096_%282%29.rar?dl=0)

You may be lucky if you flash this one and the SPIFFS is still in tact.

beicnet, 2018-10-04:
As I said before, and also DHT11 does not support negative values (it's a sensor hardware build limitation)!

Sasch600xt, 2018-10-04:
yes......i will try the DHT22 at saturday

beicnet, 2018-10-04:
All other sensor will show you full decimal / float results except DHT11!

TD-er, 2018-10-04:
OK, then we can trash this PR.
Thanks for testing/verifying it.

closed 2018-10-04 by TD-er