letscontrolit/ESPEasy #1856: ESP32

flexiti, 2018-10-04:
esp32test_1M8_partition 

At the time of saving the configuration, practically every second or third time the system falls.
In most cases, the on / off restart, but every few times the system returns to the factory settings and you have to configure everything from scratch.

TD-er, 2018-10-04:
What plugins/controllers are you using?
I have it running on several ESP32 nodes here and that clearing never happened to me, unless I made a mistake in selecting the wrong partition size.
I even have it running on a M5stack here. (PR to support m5stack will be committed later next week I hope, still some issues)
The only problems I see here are when saving plugins that use extra task settings, like Framed OLED.
Then the node crashes.

flexiti, 2018-10-05:
Probably the most thing happens during configuration: Environment - DS18b20.
I have 3 plugins: LCD2004,DS18b20 and generic info - rssi.
Display shows:
1.  time , uptime temp,
2  ssid, rssi

flexiti, 2018-10-08:
Still problem if config is saved.
e.g. once such a message has been found

> Task Device Name is not unique, conflicts with task ID #1

and

> "SaveTaskSettings taskIndex does not match"

TD-er, 2018-10-08:
The first is not a problem, but a config warning for you to use another name (unless you used an unique name, then it is a problem)

The second one should not happen.

flexiti, 2018-10-08:
all names are unique

TD-er, 2018-10-08:
Was it also unique when you first saved the plugin? (I assume you were looking in the logs)

flexiti, 2018-10-08:
Yes and not..was used before crash, I had to program everything from the beginning (there was again, config state 192.168.4.1)

The same configuration, ESP8266 works 24 hours, in the case of ESP32, it will most likely be restarted every few hours (I do not know how it will be with today's version :) )
btw. I am also  using rules like this:

```
On System#Boot do    
 event,info1
timerSet,1,4       //Resets the Timer 1 for another 4 seconds
timerSet,2,5       //Resets the Timer 2 for another 5 seconds
timerSet,3,6       //Resets the Timer 1 for another 6 seconds
endon

On Rules#Timer=1 do  
  Pulse,2,0,100
  timerSet,1,5       //Resets the Timer 1 for another 5 seconds (blue led blink)
endon

On Rules#Timer=3 do  
  event,info1
  timerSet,3,60       //Resets the Timer 3 for another 60 seconds
  Publish main/temperature/6,{"Value":[Temperature#Temperature],"No":6,"Name":"Room1"}
endon

On Rules#Timer=2 do  
  event,info1
  timerSet,2,120       //Resets the Timer 2 for another 120 seconds
endon

on info1 do
  Publish main/sonoff/35/all,{"No":35,"Relay":[Relay1#Relay1],"Temperature":[Temperature#Temperature],"Rssi":[Rssi#Rssi]}
 endon

on Switch1#Switch1 do
  if [Relay1#Relay1]=1
 GPIO,14,0

else
 GPIO,14,1

endif
endon

on Relay1#Relay1 do
 Publish main/sonoff/35/all,{"No":35,"Relay":[Relay1#Relay1],"Temperature":[Temperature#Temperature],"Rssi":[Rssi#Rssi]}
endon

on SetON do
 GPIO,14,1
endon

on SetOFF do
 GPIO,14,0
endon
```

flexiti, 2018-10-09:
esp32test_1M8_partition
Just open: Devices-DS18B20, do nothing and press save

> FS : Error while reading/writing /config.dat in 1135

flexiti, 2018-10-09:
[image]

TD-er, 2018-10-09:
OK, that sounds like a bug and probably a bug which also exists in ESP8266 code. It can also lead to crashes if there is a bug in addressing the wrong settings.

What are the position numbers in the devices list of your setup?
Which one is the Rssi and which the DS18b20 ?

flexiti, 2018-10-09:
Rssi was 2, DS18b20 was 3 ??  (I am not sure of the order, because again I have to program from 'zero'.
I will try to pay attention to it now.)

TD-er, 2018-10-09:
Adding a screenshot of the plugins present may also help indeed :)

And can you also update the topic name to reflect the actual problem? "ESP32" is a bit too generic ;)

flexiti, 2018-10-09:
ok

TD-er, 2019-02-25:
Could you test tomorrow's build on an ESP32?
I've been playing a bit with it this evening while testing a fix of handling settings and it seemed to me it was not crashing anymore when changing settings.

closed 2019-10-10 by TD-er