letscontrolit/ESPEasy #1857: Req: Display LCD2004, inversed logic checkmark for Display Button

Domosapiens, 2018-10-04:
Feature request: Add an Inversed Logic check mark possibility to the Display button.

### Expected behavior
A Display Button that can act on positive or negative logic without additional inversion HW.

### Actual behavior
The Display Button acts on negative logic: Active=On=Low (not mentioned in the Wiki!).
Movement detection sensors as PIR like HC-SR505 and Radar Doppler detector like RCWL-0516 have positive logic: Active=High.

When the detection sensor is integrated as a switch, there is an Inversed Logic check mark possibility,
but this has no influence on the Display Button.
It seems that the Display button does a direct read on the GPIO pin ???
(not bad .. that saves a Task!)

### Current work-around
A rule triggered by the detection sensor with LCDCMD,on, a timer and LCDCMD,off,
at the cost of reaction time.

fluppie, 2018-10-17:
Exactly what I want as well :). Working with a PIR sensor.

https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P023_OLED.ino
I think changes need to be in this part:

```
    case PLUGIN_TEN_PER_SECOND:
      {
        if (Settings.TaskDevicePin3[event->TaskIndex] != -1)
        {
          int index = Settings.TaskDevicePluginConfig[event->TaskIndex][0] == 0x3C
                    ? 0
                    : 1;
          if (!digitalRead(Settings.TaskDevicePin3[event->TaskIndex]))
          {
            Plugin_023_displayOn(OLED_Settings[index]);
            OLED_Settings[index].displayTimer = Settings.TaskDevicePluginConfig[event->TaskIndex][2];
          }
        }
        break;
}
```

Some inspiration in the code from the switch plugin?
https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P001_Switch.ino

```
            if (Settings.TaskDevicePin1Inversed[event->TaskIndex])
              sendState = !sendState;

            byte output_value = sendState ? 1 : 0;
event->sensorType = SENSOR_TYPE_SWITCH;
```

tonhuisman, 2022-05-06:
Feature has been added about 1 year ago via #3620

closed 2022-05-06 by TD-er