letscontrolit/ESPEasy #1859: Move Plugin 166: WiFi Power Management Functions to main repo

TD-er, 2018-10-05:
See [enesbcs/ESPEasyPluginPlayground Plugin 166: WiFi Power Management Functions](https://github.com/enesbcs/ESPEasyPluginPlayground/blob/master/_P166_WiFiMan.ino)

Heiend, 2018-10-06:
Oh yes, please. It's a nice Feature. I tested it and had no problems. I lost the dummy device values by using the modem sleep, but that is the only disadvantage I figured out.

Heiend, 2018-10-06:
I'll try it :-) I use TaskValueSet to store values in dummy devices by calling rule events via http. Than the ESP (Wemos Lolin D1 Mini Pro) goes into modem sleep and after wake up I check the dummy device values and they all where set to 0. Apart from that I can't find unexpected behavior.

Heiend, 2018-10-06:
Wait! I tested it again and now the values the values are retained. I am confused. I din't changed the Rules or other configuration. Maybe it's the best to restart the ESP after configuring the WiFiMan the first time? I don't know. Now it works like expected.