letscontrolit/ESPEasy #1860: LWT topic on OpenHAB MQTT controller

Darkhub, 2018-10-06:
Hi, I am trying to understand how ESPEasy handles LWT messages using an OpenHAB MQTT controller.

Subscribing to /ESP01/# I can see all the messages sent and if I configure a "Controller lwl topic" as "customstatus" I see that when ESPEasy boots up it:
- publishes the "LWT Disconnect Message" value to /ESP01/status/LWT
- publishes the "LWT Connected Message" value to /ESP01/customstatus (as entered in "Controller lwl topic")

If "Controller lwl topic" is left blank, both LWT messages are published to /ESP01/status/LWT

Is this the right behaviour? Or should both topics be changed according to "Controller lwl topic" value?

Thanks.

zdzichu, 2018-10-06:
From MQTT point of view, both should be send to the same topic – the one you configured. It seems ESPEasy code is wrong.

TD-er, 2018-10-06:
Based on the code in `Controller.ino`, it looks like `ControllerSettings.MQTTLwtTopic` is still empty when it sends the disconnect message to the broker when making the connection.

The same topic (`LWTTopic`) is being used to connect and to send the connected message.
Is it possible the MQTT broker still keeps the disconnect message even when the ESP node thinks it reconnects? Did you later change the "Controller lwl topic" ? Even with a number of reboots of the ESP node inbetween?

Misiu, 2019-07-12:
@TD-er is this still an issue? As @zdzichu suggested both messages (online/offline) should be sent to the same topic (/$devicename$/status)
I'd like to use ESP Easy with Home Assistant. ESP Easy will measure outside temperature and I want to know if the device is online/offline. LWT seems to be the perfect solution.
I'm going to test this next week and will get back on this for sure.

TD-er, 2019-07-12:
@Misiu I am not sure if it is still an issue.
I am looking forward to see your findings.

Misiu, 2019-07-12:
@TD-er LWT Connect and Disconnect messages would remove the need for heartbeat messages (although this would also be a nice addition, is it possible right now?)
As I mentioned I'm going to check the topics and get back to You as soon as possible

TD-er, 2019-10-27:
Is this still an issue?

Misiu, 2019-10-28:
@TD-er I moved from openHAB to Home Assistant. I recently flashed a couple of Wemos D1 and checked MQTT. I had no problems with it, but I didn't changed LWT topics. The devices displayed as offline when I unplugged Wemos from power and showed as online when the communication started again (after powering up)
The initial question was about openHAB, but since openHAB is using Homie, I'm not sure if this is still an issue. @Darkhub?

ghtester, 2019-10-28:
I believe this could be an issue with openHAB's integrated broker Moquette. In my case ESP_Easy node which  often disconnects and reconnects due to bad WiFi reception (as reported in #2684) is  probably also forcing Moquette broker to stop working in few days (filling mqttembedded.bin). To make it working again  I had to delete the mqttembedded.bin, clean the cache and restart the openHAB services and eventually reconfigure Things... So I decided to stop using integrated broker and switch to Mosquitto broker from CentOS which looks to be stable even with my "permanently" reconnecting ESP_Easy node.

jimmys01, 2019-10-28:
Openhab and lwt works great for me

TD-er, 2019-11-04:
@ghtester Since there is a new issues about the crashes and disconnects you see, I guess this one can be closed.
Please let me know if it is a different issue.

closed 2019-11-04 by TD-er