letscontrolit/ESPEasy #1863: Add Ping plugin to Mega branch

TD-er, 2018-10-07:
See [forum](https://www.letscontrolit.com/forum/viewtopic.php?p=19815#p19815)

N.B. This now uses the blocking version of ping.
It should also send an event which can be used in plugins.

lyndondr, 2018-10-24:
Any chance this will be added soon? Would there be a workaround to detect an internet connection without the ping plugin?

Grovkillen, 2018-10-24:
Just for info, related topic #950

TD-er, 2018-10-24:
@lyndondr At this moment we're working full on getting things stable and fixing bugs, so handling feature request is at a lower priority right now.

enesbcs, 2018-11-13:
How about this ping library?
https://github.com/bluemurder/esp8266-ping/blob/master/src/Pinger.cpp
As i see it is using os_timer_disarm() for waiting icmp echo, i am not sure if it is blocking function or not.

gumileeev, 2019-06-10:
Whether prompt there is a firmware already with the built-in plug-in?  I can not figure out how to embed a plugin.  Is it expected?  Please show the instructions for installing the plugin.

enesbcs, 2019-06-16:
Binary with P126 plugin:
https://www.letscontrolit.com/forum/download/file.php?id=3594

gumileeev, 2019-06-18:
> Binary with P126 plugin:
> https://www.letscontrolit.com/forum/download/file.php?id=3594

Dear enesbcs! Thank you for responding to my request for help. I could not install this file in the Sonoff Basiс. Perhaps due to the fact that this file is 4M, only 1M? Is there a similar file for 1M? Thank you very much in advance!

PR #5379 (2025-09-07): [P089] Add support for ESP32
Resolves: #4555
Resolves: #1863

Features:
- [P089] Add support for ESP32 in Ping plugin
  - Using `dvarrel/ESPping` library (with some fixes).
  - Handles the ping in a separate RTOS task, making it non-blocking.
  - Optionally shows the average ping-time to the host in msec. (`Avg_ms`)
  - Fix multi-instance use by handling actual, single-instance, ping action in a separate service that handles request sequentially.
  - Enable PluginStats for ESP32 so view trends in average ping-time.
  - Max ping-count reduced from 100 to 25 (can take up to 25 seconds in total,  [... cut]
closed 2025-09-07 by TD-er