letscontrolit/ESPEasy #1873: since 20181007 MQTT Oben Hab shows "Connection Lost"

Sasch600xt, 2018-10-09:
Hello all,

at least for me 20181007 and 20181008 shows "Connetion Lost" in MQTT Open Hab server.
But still everything is working.

First it appears iin 20181007 after 3.5 hours. With 20181008 after about 10 hours.
Once it shows "Connection Lost" it never comes back to "Connected" unless i do a reboot.

Hardware: Lolin Nodemcu V3
Image: Normal,4096

Greetz
Sascha

TD-er, 2018-10-09:
Does it need a reboot of OpenHAB/MQTT broker, or only the node itself?

And what is the reboot reason of the node when the MQTT connection is lost? (before you reboot it ofcourse)
Is the node still reachable when the connection is lost?

And what version did still work like expected?

Sasch600xt, 2018-10-09:
Hello TD-er,

- disable and enable MQTT controller worked and the node shows up as "Connected" again.

- Reset Reason | Software/System restart

- Yes, the node works perfect and is reachable

- Version before 20181007

hope that helps
Greetz
Sascha

uzi18, 2018-10-09:
have also reports that 20181006, does not work with mqtt, but have no logs, asked to check this with tomorrow build

TD-er, 2018-10-09:
What MQTT broker is being used?

uzi18, 2018-10-09:
cloudmqtt.com

TD-er, 2018-10-09:
Can you do a ping to that host from your computer to get an idea of the round trip time?
From my computer it is a stable ping time of 115 msec.
The (new introduced) default timeout is set to 100 msec in the controller.
So could you set it to 200 msec? (or about 2x the timeout you see)
I will also set the default higher for newer builds.

Sasch600xt, 2018-10-11:
hello,
right now i did an upgrade to 20181011.
I will tell in next 24 hours when the bug is gone.

Thank you
Sascha

TD-er, 2018-10-11:
Please also check the timeout setting for the controller you use.
It only sets to 300 msec as a default (when never set or when adding a new controller)

Sasch600xt, 2018-10-11:
ok.....i will, but i never changed a setting there

Sasch600xt, 2018-10-11:
i see it was 100ms for "client Timeout" i did change it to 300 now, right ?

TD-er, 2018-10-11:
Yep, and maybe also do a ping from your own computer to get an estimate of the ping times to that host you're using.

Sasch600xt, 2018-10-11:
ping is between 3ms and 68ms

TD-er, 2018-10-11:
Ah, now I see, it was @uzi18 who mentioned `cloudmqtt.com`
Still, 68 msec ping can have some higher values over the hours which may exceed 100 msec. So increase is a good idea I guess.
It was 1000 msec (default timeout), but since we're having a lot of HW watchdog reports lately, this is one of the attempts to get things more stable.

TD-er, 2019-10-27:
Is this still an issue?

TD-er, 2019-11-04:
This one is probably not an issue anymore, so I will close it.
Please let me know if it still is an issue.

closed 2019-11-04 by TD-er