letscontrolit/ESPEasy #1874: attachInterrupt() and detachInterrupt()

giig1967g, 2018-10-09:
I'm not sure if in the real world it can cause crashes, but for sure it's a bug.
I noticed that all plugins that call attachInterrupt() do not call detachInterrupt() if disabled, deleted or modified.

I noticed that in the following plugins there is a call to attach but not to detach:
003-Pulse
008-IR
046-Ventus

I am not an expert on Arduino core, but is it worth a look?

TD-er, 2018-10-09:
I have also looked a bit at that code and there is more strange about that code.
Maybe it is possible to attach multiple callback functions to a pin interrupt, but I guess not.
That leaves at least inconsistent behavior when one of these plugins is disabled/removed, so it is worth to look at.

giig1967g, 2018-10-09:
Well, I have a different idea in handling interrupts for the future (V2):
we should manage switches, pcf, pulse, MCP and all other input device with central interrupt management.
Instead of using PLUGIN_TEN_PER_SECOND, we should create an array that will contain the callback funtions for those interrupt devices.
The array will be modified everytime there is a device change.
This will free many resources that today just loop 10, 50 times per second for every device...

I am not sure I have the competence for this changes, but after stability (priority 1), rules speed (priority two) I vote for interrupt management instead of PLUGIN_TEN_PER_SECOND...

Coming back to this issue, if PLUGIN_EXIT should be called everytime the plugin is disabled or modified or deleted, I don't see these calls in the code. So probably many resources are not freed properly.
We should probably add a PLUGIN_EXIT call in all plugins that use attachInterrupt() and write detachInterrupt();

What do you think?

TD-er, 2018-10-09:
There is already an issue for that. What I want to do is to read the registers that contain all pin states at a very low level and compare that with a mask to see if one of the watched pins has changed.
This can be run very often and is less restrictive than the interrupt ones.
Don't try to use arrays for that, a 32-bit unsigned int is good enough and those can also be XOR'ed in a single instruction.

giig1967g, 2018-10-09:
ok. I dont't think have the capabilities for managing interrupts...
Just developing new features and fixing bugs is enough for me...

uzi18, 2018-10-09:
We need to look at it but for now we have problems with units not reconfigured in runtime, so problem is elsewhere.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er