letscontrolit/ESPEasy #1875: Automated binary releases

flexiti, 2018-10-09:
**Flash8266.exe**  not work under windows 10 ( at least for me)
Can't upgrade ESP8266 and ESP32 also

TD-er, 2018-10-09:
Did it work before on Windows 10?

I have also a number of strange issues regarding USB devices since the last update(s).
For example it is impossible for me since yesterday to use my Dymo label printer and I have to reconnect my USB hub quite often to be able to flash my nodes.

flexiti, 2018-10-09:
Yes, it work before (I do not remember when but the last time I used it about a month ago and then it worked)

flexiti, 2018-10-10:
if you can not actually load ESPeasy from windows 10, does that mean that no one who has windows 10 uses it? :)

flexiti, 2018-10-10:
my point is, you do not want to compile own versions, but to report errors of the recompiled version. In the set of recompiled versions there is a tool (from You) for uploading new software, but it does not work on windows 10 ... it's ok by. you ? ;)

closed 2019-10-10 by Grovkillen