letscontrolit/ESPEasy #1876: Pulse Counter Formula error

makonrus, 2018-10-09:
I was looking for ways to set the initial values ​​for a pulse counter. I decided that the best option is to use a formula like %value%+X (where X is the initial value of the counter). But the formulas do not apply! If Total = 5, then whatever formula I use is the value does not change.

### Expected behavior

Total = X, Formula = %value%+Y, Result is X+Y. (X and Y are any integers numbers)

### Actual behavior

Total = X, Formula = %value%+Y, Result still X. (X and Y are any integers numbers)

### Steps to reproduce

1. Set Total Formula = %value%+1000 (for example)
2. Total does not changed

### System configuration

Hardware:
Wemos D1 mini pro

ESP Easy version: 

Build | 20102 - Mega
Libraries | ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3
GIT version | mega-20181008
Plugins | 46 [Normal]
Build Md5 | 79b626cf6e3ad26c4c7a64a1062a4ce
Md5 check | passed.
Build time | Oct 8 2018 02:20:08
Binary filename | ESP_Easy_mega-20181008_normal_ESP8266_4096.bin

TD-er, 2018-10-09:
Can you use the same formula on any other plugin?

makonrus, 2018-10-09:
No problem with internal ADC!

TD-er, 2018-10-09:
OK, so the formula part is just broken for that specific plugin, not for all.
Good to know. (also the formula makes sense ;) )

makonrus, 2018-10-09:
It seems to me not the best idea to set the initial values ​​of the counter using the formula... I am also trying to find the answer to the question: "What if a power failure or a sudden reboot?" So far I have solved this issue with li-ion. But perhaps there are more reasonable ways to save values ​​in case of a reboot.

TD-er, 2018-10-09:
The values should be saved in RTC memory, which is restored on reboot.
As long as the device is powered, the RTC memory is kept.

If those are not restored, then that's a bug. (or the RTC data was corrupted, then it is deleted at boot)

makonrus, 2018-10-09:
those are not restored :)
But for the meter it is also important to keep the values ​​in case of power failures! Is it possible to realize this without killing the flash?

evg26, 2018-10-16:
Formulas do not work

evg26, 2018-10-16:
The formula is necessary to adjust the meter readings. The server is not enabled.

evg26, 2018-10-16:
Корректировки показаний счетчика в ESP8266 с сервера?

evg26, 2018-10-17:
[image]

Empty string

evg26, 2018-10-17:
[image]
ned labels

Grovkillen, 2018-10-17:
1mSec debounce? And the values are not stored or what do you mean? You're not giving us much to work with here.

TD-er, 2018-10-17:
I was able to reproduce the non-working formula on my test node.
What version has the emptied strings? I thought I fixed that 2 days ago.

uzi18, 2018-10-17:
looks like strcpy bug, but this should be fixed now

evg26, 2018-10-18:
mega-20181017 Empty string

TD-er, 2018-10-18:
Also after adding the right string and saving it again?

evg26, 2018-10-18:
saves changes,OK

evg26, 2018-10-18:
Formulas don't work mega-20181017

evg26, 2018-10-18:
1 pulse counter 2

TD-er, 2018-10-18:
Can you for future replies keep them in the same reply, or edit your reply to add extra information?
(as long as no newer replies are added ofcourse and the new information is within minutes from eachother)

I did not yet work on the formula part.
I have looked into it for a short while, but could not see yet why it isn't working for this specific plugin.

evg26, 2018-10-18:
Fed pulses from the button, changed the time of contact rattling from 0 to 60, the counter added 1 to 2.

evg26, 2018-10-18:
to view the meter readings you need to reload the page. You should see the meter reading changes without overloading the page. Is that possible?

TD-er, 2018-10-18:
The Device tab will reload the values using JavaScript.
But the refresh rate of that reload depends on the "interval" timer of the plugins.
The lowest set interval timer determines the time between reloads in the background.

evg26, 2018-10-29:
Analog input - internal -works. the readings are being updated.

evg26, 2018-10-29:
and the formula works

evg26, 2018-10-29:
Generic - Pulse counter it never works. Very necessary.

evg26, 2018-10-29:
GIT version: | mega-20181029
-- | --

nologin24, 2019-01-04:
I have some additional information about the problem with formulas. In the Webinterface the formulas didn't work, but for sending MQTT the formulas are calculated correctly with ESPEasy_mega-20181231.

I hope this makes it easier to find the problem.

Build: 20103  - Mega
Libraries: ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3
Plugins: 46 [Normal]

evg26, 2019-01-05:
ESPEasy_mega-20181231 formulas do not work, readings are not updated automatically, protection against contact chatter does not work.

m50434, 2019-05-25:
Is there some update on this issue?!
Unfortunately Formula stil does not work with Pulse Counters

Domosapiens, 2019-05-25:
@m50434  This works correct for me:
**%value%/396**
[image]
Debounce=0
Counter type=Delta
Mode type=Falling
Interval=60sec

mega-20190522, normal
via Domoticz controller

m50434, 2019-05-25:
Sorry, I forgot:
In my broker the values are correct in my mqtt-broker (so actually the formular works) but in esp-easy itself the values are incorrect !!!

Hudoba, 2019-07-21:
mega-20190626
calculating not working, and few version I'm tested
%valus%*10

TD-er, 2019-07-21:
> mega-20190626
> calculating not working, and few version I'm tested
> %valus%*10

And what if you use `%value%` instead of `valus%` ?

Hudoba, 2019-07-23:
Sorry for %valus% :), MQTT publish is working WEB is NOT  working. Is possible to save data to internal flash?

TD-er, 2019-07-23:
I am working on that part.
The (highly experimental) "cache controller" is storing data into .bin files on the SPIFFS.
But right now, I have not published a way to use that data again, because I'm still working on it.
The bin files are currently very simple. A sample consists of 6x32 bit, containing:
- timestamp
- some other bytes containing task ID etc.
- 4x 32 bit float value.

The reason there is no read option yet is because this will not be the final binary format.

tonhuisman, 2023-09-03:
This issue is no longer relevant, and can be closed.

TD-er, 2023-09-03:
This is still an issue, however I am not sure whether it can be fixed in a simple way.