letscontrolit/ESPEasy #1878: Switch device does not show the real state after reboot

makonrus, 2018-10-09:
### Summarize of the problem/feature request

Switch device does not show the real state after reboot. There is an "Input" boot state for this GPIO in the hardware section.

### Expected behavior

I have 3,3 Volts on the GPIO-12 input (pulled up to DC power). If i connect the input to the GND then the Device state changed from 1 to 0 (from 0 to 1 for inversed logic, that i used). If i reboot the system, then the current state of the input should be displayed (or its opposite when using inverted logic). 

### Actual behavior

There is the "0" state is displayed after rebooting. Anyway. **Also, setting or clearing the inverted logic flag does not change the displayed value in any way.** If you physically change the state of the input, then everything starts to display correctly.

### Steps to reproduce

1. Create the switch device
2. Set the inverted logic flag (I think the result will be the same for non-inverted logic)
3. Set the corresponding GPIO to "Input" boot state in Hardware section
4. Puled up the input to +3.3 V
5. Connect the input to the GND (by button for example)
6. You will see the value 1 (or 0 for non-inverted logic)
7. Rebbot you system (do not change the input state!)
8. After the reboot, you will see the value 0. It does not depend on inverted or non-inverted logic. Always 0.
9. If you now physically change the state of the input, the values ​​will be displayed correctly.

### System configuration
Hardware:
Wemos D1 mini pro

ESP Easy version:

Build | 20102 - Mega
Libraries | ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3
GIT version | mega-20181008
Plugins | 46 [Normal]
Build Md5 | 79b626cf6e3ad26c4c7a64a1062a4ce
Md5 check | passed.
Build time | Oct 8 2018 02:20:08
Binary filename | ESP_Easy_mega-20181008_normal_ESP8266_4096.bin

TD-er, 2018-10-09:
So just as a summary, the initial value of the plugin does not match the true state + settings at boot.
But as soon as you change the state on the pin itself, the correct value is set?

So it sounds like an issue with the init of the plugin?

giig1967g, 2018-10-09:
It's a known issue.
For PCF I just published a PR #1767 that fix for PCF this INIT behaviuor.
I have ready the fix for GPIO and MCP and will create a PR very soon.

makonrus, 2018-10-09:
> But as soon as you change the state on the pin itself, the correct value is set?

Yes

TD-er, 2018-10-09:
@giig1967g I don't see anything mentioned about a GPIO extender 
@makonrus This is just about the normal P001 Switch plugin with the standard GPIO pins, right?

giig1967g, 2018-10-09:
@TD-er: I know, but the same behaviour is for GPIO, PCF and MCP.
So for PCF I submitted a PR.
For the other will submit in a short time.

uzi18, 2018-10-09:
INIT issue is also related to PCF plugin

uzi18, 2018-10-09:
it is related to "Send on boot" feature, as if setted state is inverted to be sure immediately after first read found "change"

Darkhub, 2018-10-21:
I am also expecting this too, at boot the switch (configured as a push-button active low) is not initialized from the (retained) value stored on MQTT broker.

I see there are these 2 variables used in the switch plugin:
switchstate[event->TaskIndex] = Plugin_001_read_switch_state(event);
outputstate[event->TaskIndex] = switchstate[event->TaskIndex];
At least when configures ad a push-button, the switch device should subscribe to its related topic and update the states accordingly, is this possible?

giig1967g, 2018-10-22:
If I understand correctly this request, you would like to set the initial value of a switch (INPUT-type).
If my understanding is correct, then it's not possible. 
The INPUT switch read the state of the appropriate PIN but cannot set its state.
If by any chance you set a value to an INPUT-type switch it becames an OUTPUT-type switch.

If, instead, you want to set the initial value of a push-button (to keep it synchronized with another system), then it's a different story, because the push button could be set using the UserVars struct. But I have to check how to do it. The risk is that during boot the pushbutton value is set to 0, then MQTT retain changes it to 1 and this change sends an undesired event that could trigger a rule...

Darkhub, 2018-10-22:
Yes, when the switch is effectively connected to a physichal switch that's not a problem, the corresponding GPIO level is read from the pin and processed.
But, when the switch is configured as a push button, there's a sort of software flip-flop that is toggled every time the button is activated.
In this case it makes perfectly sense to check if there's an on/off state to restore on device initialization or if the corresponding topic is, for example, changed by another client posting to the same topic (just consider 2 buttons on 2 different ESPs commanding the same light, or chenging switch state via a dashboard and so on).

giig1967g, 2018-10-22:
Understood. then the label should be **feature request** and not **bug**.

TD-er, 2018-10-22:
@giig1967g If you want something like that, then we really should think about what features are needed before we start changing some code.

The way I see it, the current Switch plugin is already overloaded with features and is way too complicated to get it right.

**Input** features:
- signalling a change on a GPIO pin (on rise/fall/change, selectable)
- detect pulse width of changed GPIO pin to detect false changes (dender) and multi tap and long tap.
- send an output (at plugin level) based on changed state

**State** features:
- Retain an output state after reboot.
- option to inverse state
- option to change state depending on detected pulse width. (set duration, multitap, etc)
- override of state using commands/events (Hotel switch logic)
- send event when state is changed describing type of change (multi tap, long tap, on/off)

**Output** features:
- Set GPIO output based on state
- option to invert logic relation between state and GPIO value.
- option to set retained value after reboot.

As you can see, the features are quite easy to split into:
- input
- state
- output

And I am not sure if this should be kept all in the same plugin.
For now, the switch plugin is way too complex and should get a full redesign, since the complexity must be removed.

giig1967g, 2018-10-22:
@TD-er: I perfectly agrre. The feature requested by the OP is quite complex.
From my point it can put in stand-by.
Just remove the label "bug" as it's not a bug.

Darkhub, 2018-10-22:
So the only way to achieve such behaviour is via riles and a dummy device?

giig1967g, 2018-10-22:
it's the easiest...

closed 2019-10-11 by Grovkillen