letscontrolit/ESPEasy #1881: Hardware Watchdog

DittelHome, 2018-10-10:
I have to report a Hardware Watchdog problem, sorry.
I have 5 ESP8266 running here, only one have problems with reboot's.
Firmware: ESP_Easy_mega-20181007_normal_ESP8266_1024.bin (on all ESP's)
The problem ESP has the following devices:
- Display - LCD2004
- Generic - MQTT Import
- Environment - DHT11/12/22 SONOFF2301/7021
- Analog input - internal
- Generic - Dummy Device

In addition the follwing rule is used:

> on System#Boot do
>  timerSet,1,1
>  lcdcmd,off 
>  lcdcmd,clear
> endon
> 
> on Rules#Timer=1 do 
>  lcdcmd,clear
>  lcd,1,1,Waiting... 
>  lcd,2,1,Waiting...
>  pwm,5,200
>  TaskValueSet 5,1,200
>  lcdcmd,on
>  endon
> 
> on MQTTimport#Connected do
>  timerSet,4,3
> endon
> 
> on Time#Initialized do
>  timerSet,2,3
> endon
> 
> on LDR#Light do
>  If [LDR#Light] >550 and  [LDR#Light] <650 and [Dummy#PWM] != 20
>   pwm,5,20,3000
>   TaskValueSet 5,1,20
>  endif
>  If [LDR#Light] >650 and  [LDR#Light] <750 and [Dummy#PWM] != 40
>   pwm,5,40,3000
>   TaskValueSet 5,1,40
>  endif
>  If [LDR#Light] >750 and  [LDR#Light] <900 and [Dummy#PWM] != 50
>   pwm,5,50,3000
>   TaskValueSet 5,1,50
>  endif
>  If [LDR#Light] >900 and  [LDR#Light] <1010 and [Dummy#PWM] != 100
>   pwm,5,100,3000
>   TaskValueSet 5,1,100
>  endif
>  If [LDR#Light] >=1010  and  [Dummy#PWM] != 800
>   pwm,5,800,3000
>   TaskValueSet 5,1,800
>  endif
>  If [LDR#Light] <=550 and  [Dummy#PWM] != 1
>   pwm,5,1,3000
>   TaskValueSet 5,1,1
>  endif
> endon
> 
> on Rules#Timer=2  do
>  lcd,2,1,%sysweekday_s% %sysday%-%sysmonth%-%sysyear%     x
>  timerSet,3,2
> endon
> 
> on Rules#Timer=3  do
>  lcd,2,1,In  T:[T&H#in_temp] H:[T&H#in_hum]    x
>  timerSet,2,2
> endon
> 
> on Rules#Timer=4  do
>  lcd,1,1,Wind [MQTT_IMPORT#ws] km/h   x
>  timerSet,5,2
> endon
> 
> on Rules#Timer=5  do
>  lcd,1,1,Out T:[MQTT_IMPORT#out_temp] H:[MQTT_IMPORT#out_hum]   x
>  timerSet,6,2
> endon
> 
> on Rules#Timer=6  do
>  lcd,1,1,Rain [MQTT_IMPORT#Raintoday] l/qm   x  
>  timerSet,4,2
>  endon

The difference to the others working ESP's is the extensive use of rules, maybe thats the cause.
This ESP is using also the pwm function. In case of problem, the pwm does not work, a short timer later the esp do a reboot. After reboot, again pwm is not working.  The only why to get the esp again in working mode, ist to resore the config.dat. Very strange !!

DittelHome, 2018-10-10:
>So it seems the config file gets corrupted?

The best way to explain the error is to describe the situation in chronolic order:

- Update firmware to 20181007 from 20180712
  Result is good, ESP starts, everthing is working. After i while, the backlight of my LCD2004 is not working anymore (PWM function). ESP is rebooting periodically. 
- After some reboots, PWM fails, no backlight ! 
  The only way to restore backlight is to restore config.dat.
- Restoring config.dat, everthing works fine. After a while again backlight goes off, rebooting.
  Backlight stays off.
- Back to 20180712, also backlight not working
- restore config.dat, everthing is fine with old firmware.

The situations sounds a little bit random, therefor maybe it has smoething to do with storage overrides ? 

Sorry it sounds confusing...

DittelHome, 2018-10-10:
I read somewhere in the forum that the serial log level can be responsible for the hardware watchdogs. I should say, that was it!
Problem solved after set the serail server log level to none

closed 2018-10-10 by DittelHome

TD-er, 2018-10-10:
There are more reports on the Watchdog reboots, so even whet it isn't fixed, you can add new info to one of these topics:
https://github.com/letscontrolit/ESPEasy/projects/12

uzi18, 2018-10-10:
@TD-er good point, we can ask people to test watchdog problem with logging disabled, maybe something wrong is here.

Domosapiens, 2018-10-11:
@TD-er , @uzi18 
This one has serial line disabled (=check box Enable Serial port not active).
Is that equal to serial logging disabled?  Or is the Serial Log buffer still filled with Info level?
[image]
Watchdog comes in:
[image]
(daily updated to latest release)

Additional question:
Is the Lat/Lon setting important? These seem to be default settings?

TD-er, 2018-10-11:
I am not really sure it was correctly checking for the serial port enabled flag.
Serial Log level is not set to None in your screenshot, so it might still get filled every now and then.

I looked into it last night and concluded it was a bit more work than expected, so I made an issue for the Serial Port + logging etc.