letscontrolit/ESPEasy #1882: since 20181004 NeoPixel SK6812/WS2812B pixel errors (timing wrong ?)

amigatommy, 2018-10-10:
### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

I use a precompiled version: ESPEasy_mega_20181004_test_ESP8266_4096_VCC.
Since this version, there are problems with my Neopixel strip (SK6812 at GPIO 0) with the Plugin38.
Some LEDs light up in an uncontrolled manner. When change the color or brightness, most pixels are ok,  but some are faulty. The number and colors of the wrongly lit LED's changes. Possibly, faulty timing caused by going back from Core 2.4.2 to 2.4.1?

### Expected behavior

### Actual behavior

### Steps to reproduce

1. up to version ESPEasy_mega_20181003_test_ESP8266_4096_VCC it works perfectly
2. since version ESPEasy_mega_20181004_test_ESP8266_4096_VCC and up it don't works correctly

### System configuration

Hardware: NodeMCU V1.0, 4M Flash. NeoPixel strip with SK6812 LED's (40 Pixels in use)
                 Power Supply 5V/5A, 1000uF Elko directly on strip supply wires 

ESP Easy version: ESPEasy_mega_20181004_test_ESP8266_4096_VCC

ESP Easy settings/screenshots: 

### Rules or log data

TD-er, 2018-10-10:
Can you build a version yourself, to verify it may be related to the core library?
Or should I build one for you?

amigatommy, 2018-10-11:
Can you could build a version for me please ?

TD-er, 2018-10-21:
Is this still an issue with the latest builds?
Maybe we should make a split in builds between core 2.4.1 and 2.4.2

TD-er, 2019-10-27:
Not replied for over a year, so I consider it fixed.

closed 2019-10-27 by TD-er