letscontrolit/ESPEasy #1883: esp32test_1M8_partition - device config save

flexiti, 2018-10-10:
Configuration as on screen shots, after the day of work the device configuration is opened, only pressed Save and effect as on the next picture

[image]
[image]

TD-er, 2018-10-10:
Hmm, that sounds like a (serious) bug. I will look into these routines to see where it may end up with a wrong value.

flexiti, 2018-10-16:
@TD-er 
I have hanged core_esp32_1_4_0.platform to core_esp32_1_3_0.platform and all looks more stable.
Did not I read something? how do you compile?

tonhuisman, 2022-04-18:
`This issue never got any update, but AFAICS, it is no longer valid, so can be closed.

closed 2022-04-18 by TD-er