letscontrolit/ESPEasy #1888: Brightness value not set correctly for P042 NeoPixel Candle

TD-er, 2018-10-11:
See [forum](https://www.letscontrolit.com/forum/viewtopic.php?p=32211#p32211)

Suggested fix also adds a number of effects and makes the number of LEDs selectable.

ozett, 2018-10-11:
changing number of leds needs afterwards hard-reboot. (this should be mentioned explicitly)
or coded as a pop-up or hint?

ozett, 2018-10-11:
as i have a 5m stripe running, it is not a candle anymore...
maybe some renaming or merge with the general neopixel?

this plugin works great with effects on stripes. maybe i should post a picture in the forum?

TD-er, 2018-10-11:
Pictures/animations are always nice to see :)
It is a good suggestion to merge plugins when they are much alike.

Why does it need a reboot when changing nr LEDs?

ozett, 2018-10-11:
dont know what was there originally coded.
maybe the lib initializes the led-count only at boot?

but if i increase the led-count in the webform, no more leds light up immediately.
only after a reboot the increased leds are lighten up.

should i test anything?

TD-er, 2018-10-11:
I will have a look at the code to see what is happening.
Last night it was already past 2am and I just looked at the changes for about 5 minutes when creating this issue.

ozett, 2018-10-11:
ok, i am no coder, but willing to help as much as i can..
;-)

ozett, 2018-10-27:
"wheel"-statement in this patched version conflicts with another development -plugin
-> https://github.com/djcysmic/NeopixelBusFX/issues/5

maybe change "wheel" here in the source?

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er