letscontrolit/ESPEasy #1891: Generic UPD controller - send only first value from sensor

jezikk82, 2018-10-11:
Hi,

in the newest official build (Release mega-20181011) the is a problem with populating date by the controller. It only sent firs value from sensor. 

Details:
Sensor: BME280 or TSL2561 or SDS011
Controllers - Generic UDP
Controller Publish: %sysname%_%tskname%_%valname%=%value%

Only first value from each of sensor is sent instead of 2 or 3 shown in UI. 
Previously on mega-20180126 works fine.
Problem confirmed on updated and new unit.

TD-er, 2018-10-11:
Can you give a screenshot of the controller settings?
And please also press "save" on the settings page to check if it is still a problem.
Do you also have an example/screenshot/log entry from the older version to show what the UDP data was sent before.

jezikk82, 2018-10-11:
See screenshot below:
[image]
I dont have anything from old versions. I can share some statistics from Loxone (the reciever of the populated UDP data). Unfortunately, It'll be one by one value.

TD-er, 2018-10-11:
The only thing that really has changed is the check of the result value from `portUDP.endPacket()`
See the code here:
https://github.com/esp8266/Arduino/blob/d17ffc287493e76e7d6895c5c623ff4ae66baaa6/libraries/ESP8266WiFi/src/WiFiUdp.h#L70-L72

I've looked at other calls to  `endPacket()` and the return value is never checked.
So I guess the return value check should be removed.

jezikk82, 2018-10-11:
What I see is that the package with the same value is sent 10-12 times. Always it's the firs value from the sensor. 
In the mentioned code is only part of building udp package, but I think the problem is in getting values from sensor.

uzi18, 2018-10-11:
and sensor configuration?

jezikk82, 2018-10-11:
E.g: [image]

Config is exactly the same as in version from Jan.

TD-er, 2018-10-11:
It is related to the part of code I showed.

In short, some controllers only send a single value at once.
So the values are (now) collected at once and sent in sequence to the receiving end of the controller.
If that fails, it will retry for N times (set in the controller settings) and if that fails, it will throw away the entire set of samples taken at a certain moment (thus the other values from that sensor).

It decides it failed by that test I posted.
So I will now remove that check and thus for now make the number of retries for that controller useless.
N.B. attempts without wifi connection active will not be registered as attempt.

jezikk82, 2018-10-12:
Yesterday I also tried the generic http controller and problem is exactly the same. Only firs value sent. 

I know that values are sent one by one, my receiver log all incoming calls and  I can confirm that there are no connectivity problems because everything else works fine. 
Number of resent packages is not connected with the max reties value. I checked it and always it's 11-12 packages received (ever if I have retries set to 5 and 15).   

I created dummy devices with single values and everything is sent correctly, only one time. Repetitions only occur for sensors/dummmy device that have more them one value set. 

**Temporary workaround** is to create dummmy devices for every value we want to sent out, fill them in with rules and sent by controller.

uzi18, 2018-10-12:
Try generic http advanced controller, it has capability to send bunch of data

jezikk82, 2018-10-12:
Sorry, I dont have backend to work with batch data and I dont have time to prepare/change env. 
For know I posted workaround which works, only limitation is number of 12 devices. 
In case you use HTTP GET notification you don't have that problem because you can sent this data by rules call.

TD-er, 2018-10-12:
Since I'm already working on the same bug for Generic UDP, I will also look into this one.
Does your receiving end reply with an acknowledgement?
Generic HTTP controller waits for an acknowledgement.
But if the receiving end is not sending an acknowledgement, the check for result will fail.

So maybe I should add a checkbox to the controller settings to explicitly check return values or ignore them.

TD-er, 2018-10-12:
The check for result is currently also used for single value plugins, but worst-case scenario is the same is sent a number of times, but it will not be noticable in most systems which log the values.

stefbo27, 2018-10-15:
For me the UDP multivalue problem still exisits (a BME280 only sends temperature). In the log-file I see that all messages are created correctly, however only the first (Temperature) is sent (But 3 times). In the loop over all values  of CPLUGIN_PROTOCOL_SEND (line51ff) the element.txt array is populated for element 0 to 2 (in the case of 3 values).  In the do_process...queue method only the element.valuesSent element is sent. For my config valuesSent stays at 0. It should be incremented by checkDone but never is. Strange.... I uncommented the MustCheckReply so that the method always returns with checkDone(true);

TD-er, 2018-10-15:
I will look into this tomorrow and setup some UDP receiver to be able to test it myself.

jezikk82, 2018-10-16:
Unfortunately, problem is still not fixed. Tested with yesterday release.
In my case Temp value of BME280 is sent usually 10 time, even though I have set 5 tries if no ACK.

uzi18, 2018-10-16:
@jezikk82 please set logging level to ERROR and paste log here

jezikk82, 2018-10-18:
183791: EVENT: Clock#Time=Thu,06:43
183807: EVENT: Clock#Time=Thu,06:43 Processing time:16 milliSeconds
184011: UDP  : C010 connecting to 192.168.1.240:7000
184241: UDP  : C010 connecting to 192.168.1.240:7000
184465: UDP  : C010 connecting to 192.168.1.240:7000
184687: UDP  : C010 connecting to 192.168.1.240:7000
184908: UDP  : C010 connecting to 192.168.1.240:7000
185008: WD   : Uptime 3 ConnectFailures 0 FreeMem 11824
185129: UDP  : C010 connecting to 192.168.1.240:7000
185350: UDP  : C010 connecting to 192.168.1.240:7000
185912: SensorSendTask()     stats: Count: 9 Avg/min/max 103592.78/4140/362402 usec
185913: sendData()           stats: Count: 7 Avg/min/max 67773.43/20297/91697 usec
185913: Compute formula      stats: Count: 7 Avg/min/max 19.57/14/23 usec
185913: setNewTimerAt()      stats: Count: 2066 Avg/min/max 347.93/299/461 usec
185914: Delay queue     C010 stats: Count: 18 Avg/min/max 2818.28/2646/3004 usec
185914: timeDiff()           stats: Count: 988166 - CPU cycles per call: 8.57
185914: Scheduler stats: (called/tasks/max_length/idle%) 253652/2042/17/64.20
186015: UDP  : C010 connecting to 192.168.1.240:7000
186240: UDP  : C010 connecting to 192.168.1.240:7000
186465: UDP  : C010 connecting to 192.168.1.240:7000
186687: UDP  : C010 connecting to 192.168.1.240:7000
186910: UDP  : C010 connecting to 192.168.1.240:7000
187138: UDP  : C010 connecting to 192.168.1.240:7000
187359: UDP  : C010 connecting to 192.168.1.240:7000
187585: UDP  : C010 connecting to 192.168.1.240:7000
187807: UDP  : C010 connecting to 192.168.1.240:7000
188029: UDP  : C010 connecting to 192.168.1.240:7000
188252: UDP  : C010 connecting to 192.168.1.240:7000
188476: UDP  : C010 connecting to 192.168.1.240:7000
188698: UDP  : C010 connecting to 192.168.1.240:7000
188919: UDP  : C010 connecting to 192.168.1.240:7000
189145: UDP  : C010 connecting to 192.168.1.240:7000
189367: UDP  : C010 connecting to 192.168.1.240:7000
189589: UDP  : C010 connecting to 192.168.1.240:7000
189810: UDP  : C010 connecting to 192.168.1.240:7000
196548: EVENT: BME280#Pressure=1016.17
196559: EVENT: BME280#Pressure=1016.17 Processing time:11 milliSeconds
196581: Meteo_BME280_Temperature=24.2
196584: Meteo_BME280_Humidity=42.4
196587: Meteo_BME280_Pressure=1016.2
196626: UDP  : C010 connecting to 192.168.1.240:7000
196642: Command: taskvalueset
196644: Command: taskvalueset
196849: UDP  : C010 connecting to 192.168.1.240:7000
197070: UDP  : C010 connecting to 192.168.1.240:7000
197291: UDP  : C010 connecting to 192.168.1.240:7000
197517: UDP  : C010 connecting to 192.168.1.240:7000
197743: UDP  : C010 connecting to 192.168.1.240:7000
197965: UDP  : C010 connecting to 192.168.1.240:7000
198187: UDP  : C010 connecting to 192.168.1.240:7000
198408: UDP  : C010 connecting to 192.168.1.240:7000
198629: UDP  : C010 connecting to 192.168.1.240:7000
198850: UDP  : C010 connecting to 192.168.1.240:7000
215008: WD   : Uptime 4 ConnectFailures 0 FreeMem 12064
215918: Plugin call 10 p/s U stats: Count: 298 Avg/min/max 1126.57/768/1499 usec
215918: Plugin call  1 p/s   stats: Count: 30 Avg/min/max 5773.93/5477/9418 usec
215919: SensorSendTask()     stats: Count: 4 Avg/min/max 121814.75/4503/351564 usec
215919: sendData()           stats: Count: 1 Avg/min/max 123521.00/123521/123521 usec
215919: Compute formula      stats: Count: 1 Avg/min/max 18.00/18/18 usec
215920: setNewTimerAt()      stats: Count: 2183 Avg/min/max 348.84/299/447 usec
215920: Delay queue     C010 stats: Count: 29 Avg/min/max 2835.55/2548/3330 usec
215920: timeDiff()           stats: Count: 1018699 - CPU cycles per call: 8.57
215921: Scheduler stats: (called/tasks/max_length/idle%) 259991/2152/17/64.60
241233: Meteo_Lux_Lux=0.0
241236: Meteo_Lux_Infrared=0
241239: Meteo_Lux_Broadband=0
241284: DS   : Temperature: 23.56 (28-a4-ed-b7-8-0-0-9a)
241287: EVENT: Temp_DS18B20_board#Temperature=23.56
241299: EVENT: Temp_DS18B20_board#Temperature=23.56 Processing time:12 milliSeconds
241324: UDP  : C010 connecting to 192.168.1.240:7000
[... 251 lines cut ...]
395918: Delay queue     C010 stats: Count: 29 Avg/min/max 2822.97/2583/3223 usec
395919: timeDiff()           stats: Count: 1009714 - CPU cycles per call: 8.57
395919: Scheduler stats: (called/tasks/max_length/idle%) 257164/2147/17/64.50
421237: Meteo_Lux_Lux=0.0
421240: Meteo_Lux_Infrared=0
421243: Meteo_Lux_Broadband=0
421285: DS   : Temperature: 23.56 (28-a4-ed-b7-8-0-0-9a)
421288: EVENT: Temp_DS18B20_board#Temperature=23.56
421301: EVENT: Temp_DS18B20_board#Temperature=23.56 Processing time:13 milliSeconds
421326: UDP  : C010 connecting to 192.168.1.240:7000
421547: UDP  : C010 connecting to 192.168.1.240:7000
421768: UDP  : C010 connecting to 192.168.1.240:7000
423001: Meteo_Dust_PM2.5=11.7
423004: Meteo_Dust_PM10=15.1
423024: EVENT: SysTime#SysTime=0.00
423037: EVENT: SysTime#SysTime=0.00 Processing time:13 milliSeconds
423041: EVENT: SysTime#=0.00
423407: Meteo_BME280_Hum_=0.00
423410: Meteo_BME280_Hum_=0.00
423416: Command: taskvalueset
423807: UDP  : C010 connecting to 192.168.1.240:7000
423817: EVENT: Clock#Time=Thu,06:47
423829: EVENT: Clock#Time=Thu,06:47 Processing time:12 milliSeconds

Only 4 min tracking.

TD-er, 2018-10-18:
Have you set the "Check Reply" to "Ignore Acknowledgement" ?

uzi18, 2018-10-18:
please try to. disable. rules for a. moment and check if it is working ok.

Tom-Bom-badil, 2018-10-19:
Nope, disabling rules didn't help. Having same isse, see [#1.892](https://github.com/letscontrolit/ESPEasy/issues/1892#issuecomment-431509455).
/tom

TD-er, 2018-10-20:
@stefbo27 
The general idea of this controller "delay queue" system is as follows:

- Gather all values (of a plugin) in one take
- send to the receiving end of the controller with at least the minimum send interval from the last sent message.
- If all values of a sample are sent, remove it from the queue and start with the next in line.

This last item is a bit of an issue for the controllers which can only send a single value in a message.
For the few controllers which have the `checkDone` function, the sample will be processed until all values are sent.
So it is meant to be kept in the queue for as long as all values are being sent.

See this part of the (macro) code: 
https://github.com/letscontrolit/ESPEasy/blob/0afab53659176167cb85faf7dce5ca0f71a9021b/src/_CPlugin_Helper.h#L325-L352

And for completeness, the `checkDone` function:
https://github.com/letscontrolit/ESPEasy/blob/0afab53659176167cb85faf7dce5ca0f71a9021b/src/_CPlugin_Helper.h#L116-L119

This `valuesSent` variable is being used here:
https://github.com/letscontrolit/ESPEasy/blob/0afab53659176167cb85faf7dce5ca0f71a9021b/src/_C010.ino#L78-L101

So the only thing I can think of why this doesn't work, could be the actual `valuesSent` is not updated, but just a copy (not a reference) to the same object.

TD-er, 2018-10-20:
Hmm, it looks like it is indeed working on a copy instead of the object in the list.
So I will try to change that.
This will also result in less memory usage, since the object doesn't have to be copied.

TD-er, 2018-10-20:
Should be fixed by PR #1926

Tom-Bom-badil, 2018-10-20:
Thanks TD-er, will check as soon as nightly build is online.
/tom

Tom-Bom-badil, 2018-10-21:
Yes, seems fixed - all values are transfered to the controller properly.
Thanks again for all your good work! :)
/tom