letscontrolit/ESPEasy #1892: Environment - DHT11/12/22 SONOFF2301/7021" only sends the first valuename

fly74, 2018-10-11:
Hi Developers!,

since a while, Module "Environment - DHT11/12/22 SONOFF2301/7021" only sends the first valuename to the controller. temp is sent and hum not. Can anybody confirm it or have I done anything wrong?

Greets fly74

TD-er, 2018-10-11:
What controller are you using?
See also #1891 , #1835

fly74, 2018-10-12:
It's a Generic HTTP. In the past the Controller makes 2 separate Posts to the target.

[image]

fly74, 2018-10-12:
Firmware
--
Build | 20102  - Mega
Libraries | ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 1.4.0-RC2
GIT version | mega-20181011
Plugins | 46 [Normal]
Build Md5 | e3ddee5528cf7995786ff12deb62425
Md5 check | passed.
Build time | Oct 11 2018 02:18:28
Binary filename | ESP_Easy_mega-20181011_normal_ESP8266_4096.bin

TD-er, 2018-10-12:
What version did work?
Probably over 2 months ago?

TD-er, 2018-10-14:
Have you checked the current set value of the "Check Reply" value?
It is a new value, so not entirely sure if it is using the new default.
Maybe also try to save the controller settings.

fly74, 2018-10-14:
It double Posts the temp value at the webserver:

[image]

my config:
[image]

[image]

TD-er, 2018-10-14:
Check Acknowledgement is still set, so set it to "ignore..."

fly74, 2018-10-14:
Checked both values with "Check Reply". Created a new controller and new device, still 2 posts with temp.

fly74, 2018-10-17:
Rolled back to mega-20180904, what was the first in nearest Releases what posts both values.

Tom-Bom-badil, 2018-10-19:
I can confirm the bahvior, but it used to work until I upgraded to mega-20181011. Version before was from ~August this year.

Having 3 sensors for calibration/testing, all connected to the same NodeMCU:

- SHT31 (temp/hum),
- BME280 (temp/hum/press),
- BMP180 (temp/press).

All 3 sensors read all values and display them in the web interface, but for each of them only temperature gets transferred to the controller (through Generic UDP). As said, it used to work properly until the recent upgrade.

@fly74: Thanks for the hint on the last 'good' version, will give it a try.

/tom

TD-er, 2018-10-20:
Should be fixed by PR #1926

fly74, 2018-10-21:
confirmed, mega-20181021 is working properly again. Thanks!

closed 2018-10-21 by fly74