letscontrolit/ESPEasy #1895: Bug in I2C.ino...

v-a-d-e-r, 2018-10-12:
//**************************************************************************/
// Reads a 32 bit value starting at a given register over I2C
//**************************************************************************/
int32_t I2C_read32_reg(uint8_t i2caddr, byte reg) {
  int32_t value;

  Wire.beginTransmission(i2caddr);
  Wire.write((uint8_t)reg);
  Wire.endTransmission(END_TRANSMISSION_FLAG);
  Wire.requestFrom(i2caddr, (byte)3);  <<<< Should this not be a 4 bytes request?
  value = (((int32_t)Wire.read()) <<24) | (((uint32_t)Wire.read()) << 16) | (Wire.read() << 8) | Wire.read();

  return value;
}

TD-er, 2018-10-12:
I think you're right. I have to look into the Wire documentation, but I think you're right.

v-a-d-e-r, 2018-10-12:
It looks like a typical copy'n'paste fault from the code above... ;-)

TD-er, 2018-10-13:
Should be fixed now.

closed 2018-10-13 by TD-er