letscontrolit/ESPEasy #1896: Increase CPU speed to 160MHz

giig1967g, 2018-10-12:
_Originally posted by @micropet in https://github.com/letscontrolit/ESPEasy/issues/1774#issuecomment-424992609_

Hi,
I wonder why the nightly builds aren't build with CPU speed at 160MHz. Are there some limitations or risks using 160MHz?

micropet, 2018-10-12:
Hi Plebs.

 

The last versions run absolutely perfectly at 160 MHz.

By now I have online times of 14 days.

 

At 80 MHz, of course, you can see design errors faster.

 

There is no risk or restriction.

 

Greetings Peter

 

Originally posted by @micropet <https://github.com/micropet>  in #1774 (comment) <https://github.com/letscontrolit/ESPEasy/issues/1774#issuecomment-424992609>

giig1967g, 2018-10-12:
> 
> 
> @giig1967g
> is the plugin P022 "Extra IO - PCA9685" from you?

no. it's not.

giig1967g, 2018-10-12:
> 
> The last versions run absolutely perfectly at 160 MHz. By now I have online times of 14 days. At 80 MHz, of course, you can see design errors faster. There is no risk or restriction. Greetings Peter

But then why we don't set the 160MHz speed by default in the nightly builds?

TD-er, 2018-10-12:
I've read the reports/discussions about the success stories of running at 160 MHz.
Then I became curious and started reading about it.

To be honest I have not found explicit statements whether it is an official supported speed.
And there are reports of people not able to move back again. I'm not sure if these are reports related to other issues or actually related to speed. Also older libraries may have changed things in the past they shouldn't have.

And at the moment we're facing enough stability issues, so we don't want to add new possible causes.

uzi18, 2018-10-12:
maybe we can add checkbox with 80/160 MHz witch defaults to 80?

uzi18, 2018-10-12:
it is possible to change during runtime, check new plugin with sleep. support here

v-a-d-e-r, 2018-10-13:
@TD-er: 160MHz is officially supported. From the data sheet:
.... The CPU clock speed is 80MHz. It can also reach a maximum value of 160MHz. ....