letscontrolit/ESPEasy #1897: BME680 - using bosch lib - IAQ stuck at 25

stoffelll, 2018-10-13:
### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

Hi,

I'm still trying to use the Bosch lib for reading values of the BME680 (precisly to get the IAQ calculated by the BOSCH lib). Unfortunately the IAQ is stuck at 25 (and IAQ-Accuracy is at 0), which is indicating that the sensor is still in calibration phase. Temperature, Pressure and Humidity is working fine.

With the original Bosch "basic" example, the sensor is running perfectly (https://github.com/BoschSensortec/BSEC-Arduino-library/tree/master/examples/basic).

I already opened a ticket https://github.com/BoschSensortec/BSEC-Arduino-library/issues/25 but I'm not getting answers. Maybe you have an idea what I'm doing wrong in the ESP Easy plugin.

### Expected behavior

Finish calibration and provide correct IAQ (not always 25)

### Actual behavior

IAQ is always 25. means it is always in calibration

### System configuration

Hardware: NodeMCU v3 and BME680

ESP Easy version:  2018-09-22

Thanks for your support!
SL

matthew73210, 2018-11-04:
Any luck ? I'll give your latest code a try and post the results !

stoffelll, 2018-11-04:
No unfortunately not. 

I'll added later (in a few hours) my latest code, as I'm using the latest Lib in the mean while.

Thanks
SL

TD-er, 2018-11-04:
Please take into account the license of the Bosch library when posting it.
Officially one must sign an agreement on the Bosch website before download of the (pre-compiled) library
It is probably not allowed to distribute a pre-compiled firmware binary including this Bosch library, so we have to look into that.

matthew73210, 2018-11-04:
Thanks @TD-er didn't think about that..
I have a version based on bluedots binarys and adafruits mqtt libs. Works a charm but a bit of a mismash. 

I'll give it a shot @stoffelll .

stoffelll, 2018-11-08:
@matthew73210: Did you tried it already?

THanks!

stoffelll, 2018-12-05:
@matthew73210: any news?

uzi18, 2018-12-05:
I think it is possible to count IAQ without this lib, but have no such sensor currently.

stoffelll, 2018-12-05:
Yeah it is possible, but not that accurate. In addition you can get more values as just the IAQ with the bosch Lib.

roger-, 2020-01-26:
Any updates? Does this work?

TD-er, 2020-05-02:
As an update for future implementation, see [these findings on the forum](https://www.letscontrolit.com/forum/viewtopic.php?p=44475#p44475)